microsoft/AirSim #2822: I would like to see the three cameras in my unreal environment

Jean-MichMich, 2020-07-01:
Hi everyone,

I've a little problem in my unreal environment. I builded Airsim and I created a custom map with the Airsim plugin, but when I press play, I can see the drone of course but I can't see the three cameras attached to it...

I searched a lot of solutions on internet but it didn't work each time. 

Thanks in advance for all your responses ! :)

LSBOSS, 2020-07-01:
Hi @Jean-MichMich ,

here are three things you can check: 

* Make sure the GameMode of your custom map ist set to AirSim. See World Settings for details:
[image]

* After pressing play, and flying around with your drone, try pressing 1, 2 and 3 on your keyboard to see depth, segmentation and rgb sub windows (they are configured by default)

* If you have attached your own virtual cameras to your drone you need to configure the corresponding SubWindow entry in your settings.json: 
```json
"SubWindows": [
    {"WindowID": 0, "MyFirstCustomCamera": "0", "ImageType": 0, "Visible": true},
    {"WindowID": 1, "MySecondCustomCamera": "0", "ImageType": 0, "Visible": true},
    {"WindowID": 2, "MyThirdCustomCamera": "0", "ImageType": 0, "Visible": true}
  ],
```

More details here: https://github.com/microsoft/AirSim/blob/master/docs/settings.md#subwindows

Jean-MichMich, 2020-07-03:
Hi @LSBOSS !

Thank you so much it worked for me ! i just forgot to update my settings.json !! ^^'

closed 2020-07-03 by Jean-MichMich