microsoft/AirSim #2824: How to change the directory of settings.json?

aaronalmeida, 2020-07-02:
I want to look for the settings.json file within the same folder where the unreal engine .uproject file is located rather than looking for it within ../Documents. I have been struggling trying to find the where and what to edit in the code. 

Thanks!

rajat2004, 2020-07-03:
See #2668

aaronalmeida, 2020-07-06:
Hmmm, I tried what #2668 mentioned as well as #715. I couldn't seem to get it working myself.  

According to #715, if a settings.json is located in the same directory of the .exe, it should prioritize that first. In my case it picked the ../document/settings.json right away regardless. 

I also tried giving the location of the directory where the settings.json should be looked for. Again, it just went straight to ../document/settings.json. 

[image]

Not sure how to proceed!

mhl787156, 2020-07-15:
I don't know if you solved the problem, but what I found is that you can either (1) Default load settings.json from Documents, (2) pass the json as a string into --settings. The option that is mentioned in #715 with default loading if settings.json is in the same directory as the exe doesnt seem to work! 

Therefore what I do is have a python script construct the settings json and launch my version of AirSimProject.exe:
AirSimProject.exe --settings "{ ... my json file here, i.e. json.dumps(settingsjson) ...}"
Also note that you may have to add extra escape characters into the json string. This way you can have 'dynamic' settings as well! 

Hope this helps someone :)

aaronalmeida, 2020-07-15:
Hey! I actually was not able to solve it :( And I'm glad I'm not the only one who noticed #715 doesn't work! The python solution you mentioned sounds awesome. Do you mind showing a part of your python code that achieves this? I tried running:

`AirSimProject.exe  --settings "{\"SettingsVersion\": 1.0, \"SimMode\": \"Car\"}"` 

in command prompt which is the basic command mentioned in #715 and which seems to be working for you as well. Unfortunately I get this error when I run it and the application opens up:
[image]

Appreciate the input and help!

mhl787156, 2020-07-15:
Hey! Yeah I just ended up looking directly at source! I think you are very close, but are falling prey to damn escape characters on windows! You also need to escape the backslashes I think so you end up with \\\" for each quote mark. I basically use this class as my handler to start and close the (built/ cooked) UE4 program too: 
```python
class UE_Game_Handler(object):
    """
    Available Levels:
        - ApartmentLevel
        - BridgeLevel
        - CityLevel
        - TurbineLevel
    """

    def __init__(self,
                 level:str,
                 executable_path:str,
                 settings_dict,
                 log=False):
        self.level = level
        self.executable_path = executable_path
        self.settings_dict = settings_dict
        self.process = None

        folder_path = os.path.join(tempfile.gettempdir(), "airsim_drone")
        try:
            os.makedirs(folder_path)
        except OSError:
            if not os.path.isdir(folder_path):
                raise
        self.log = log

    def __enter__(self):
        jsonstring = json.dumps(self.settings_dict)
        jsonstring = jsonstring.replace('"', '\\\"')
        # logging_file = self.logging_file.replace('"', '\\\"')
        self.process = subprocess.Popen(
            [self.executable_path,
             '-log' if self.log else '',
             self.level,
             '--settings', jsonstring]
        )
        print(f'UE4 Process {self.process.pid} Started')
        return self

    def __exit__(self, type, value, traceback):
        if os.name == 'nt':
            returncode = subprocess.call( # Kill Process on windows
                    ['taskkill', '/F', '/T', '/PID',  str(self.process.pid)]
                )
        else:
            self.process.terminate()
            returncode = self.process.wait()
        print(f'UE4 Process {self.process.pid} Closed')

        return False # propogate any exceptions
```
Essentiall note the __enter__ function where I call replace - hope that helps!

aaronalmeida, 2020-07-15:
This is so helpful thank you so much!! I appreciate it :D

lucafei, 2020-10-27:
> Hmmm, I tried what #2668 mentioned as well as #715. I couldn't seem to get it working myself.
>
> According to #715, if a settings.json is located in the same directory of the .exe, it should [...]

hello, i'm also doing this, can you explain how to solve this? and where can the AirSim.exe be found?