microsoft/AirSim #2827: Default camera position front_right and front_left

4iue0, 2020-07-04:
My English skill is poor, so it is difficult to read it.

Please tell me default camera position.
 I want to get distance from object to drone.

rajat2004, 2020-07-07:
You can get info about camera using `simGetCameraInfo()` API, see [cv_mode.py](https://github.com/microsoft/AirSim/blob/master/PythonClient/computer_vision/cv_mode.py) for example usage. As to the position of `front-left`, `front-right` cameras from the output of `cv_mode.py` -

`front-left` (Camera no. 1)- 
```
'position': <Vector3r> {   'x_val': 0.0,
    'y_val': 0.125,
    'z_val': -0.0}},
```

`front-right` (Camera No. 2) -
```
'position': <Vector3r> {   'x_val': 0.0,
    'y_val': -0.125,
    'z_val': -0.0}},
```

4iue0, 2020-07-07:
@rajat2004 
Thank you for your advise!
My problem was solved!

closed 2020-07-07 by 4iue0