microsoft/AirSim #2828: "DisparityNormalized" is always black image

4iue0, 2020-07-04:
My English skill is poor, so it may be difficult to read it.

I use client.simGetImages to get DisparityNormalized image.
However I always get just black  #image.
[image]

I use this python code
**responses = client.simGetImages([
    airsim.ImageRequest("1", airsim.ImageType.DisparityNormalized)])
    print('Retrieved images: %d' % len(responses))
    tmp_dir = os.path.join(tempfile.gettempdir(), "airsim_drone")
    print ("Saving images to %s" % tmp_dir)
    try:
        os.makedirs(tmp_dir)
    except OSError:
        if not os.path.isdir(tmp_dir):
            raise
    for idx, response in enumerate(responses):

        filename = os.path.join(tmp_dir, str(idx))

        if response.pixels_as_float:
            print("Type %d, size %d" % (response.image_type, len(response.image_data_float)))
            airsim.write_pfm(os.path.normpath(filename + '.pfm'), airsim.get_pfm_array(response))
        elif response.compress:
            print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
            airsim.write_file(os.path.normpath(filename + '.png'), response.image_data_uint8)
        else:
            print("Type %d, size %d" % (response.image_type, len(response.image_data_uint8)))
            img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) 
            img_rgb = img1d.reshape(response.height, response.width, 3)
            cv2.imwrite(os.path.normpath(filename + '.png'), img_rgb) **

EdwinMeriaux, 2020-07-05:
@4iue0 
Hello!
I am not very familiar with the disparity normalized images but I I've used the image API a lot recently so I may be able to help.
So are you not trying to set the pixels_as_float = true in your code. I checked on the API docs:
https://github.com/microsoft/AirSim/blob/master/docs/image_apis.md

"You normally want to retrieve disparity image as float (i.e. set pixels_as_float = true"

(I'm not 100% sure)I think there is an example of that here:
https://github.com/microsoft/AirSim/blob/master/Examples/DataCollection/StereoImageGenerator.hpp

on line 66:
"ImageRequest("1", ImageType::DisparityNormalized, true)"

4iue0, 2020-07-07:
@EdwinMeriaux 
Thank you for your advise !
My problem was solved.

closed 2020-07-07 by 4iue0

EdwinMeriaux, 2020-07-09:
@4iue0 

Awesome!
Was what I said the solution?