microsoft/AirSim #2829: build problem

pengzhi1998, 2020-07-04:
Hi everyone, I have a problem when building UnrealEngine following [this](https://microsoft.github.io/AirSim/build_linux/).
When I run this command: `./GenerateProjectFiles.sh`, the following problem happens:
```
ERROR: Unable to instantiate instance of 'BenchmarkToolTarget' object type from compiled assembly 'UE4ProgramRules'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> UnrealBuildTool.BuildException: GetBuildPlatform: No BuildPlatform found for Linux
         at UnrealBuildTool.UEBuildPlatform.GetBuildPlatform (UnrealBuildTool.UnrealTargetPlatform InPlatform, System.Boolean bInAllowFailure) [0x0003e] in <c8c22d78b4f54e46a68f2fef5a965dbe>:0 
         at UnrealBuildTool.TargetRules..ctor (UnrealBuildTool.TargetInfo Target) [0x002fd] in <c8c22d78b4f54e46a68f2fef5a965dbe>:0 
         at BenchmarkToolTarget..ctor (UnrealBuildTool.TargetInfo Target) [0x00000] in <7d89c5f175824ae6bbe453be27c8e90a>:0 
         at (wrapper managed-to-native) System.Reflection.MonoCMethod.InternalInvoke(System.Reflection.MonoCMethod,object,object[],System.Exception&)
         at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00005] in <98fac219bd4e453693d76fda7bd96ab0>:0 
          --- End of inner exception stack trace ---
         at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00017] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0007a] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.Reflection.MonoCMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0002f] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at UnrealBuildTool.RulesAssembly.CreateTargetRulesInstance (System.String TypeName, UnrealBuildTool.TargetInfo TargetInfo) [0x00093] in <c8c22d78b4f54e46a68f2fef5a965dbe>:0
```
And I have tried several times, the same problem happens all the time.
I'm installing it on a remote server. (On my laptop, same problem happens) Its OS is Ubuntu 18.04. And the version of UnrealEngine is 4.24. Is there anyone who has faced the same problem? 
Thank you for your help!!

pengzhi1998, 2020-07-05:
Fixed this, the problem came from that the downloading of toolchain went wrong.

closed 2020-07-05 by pengzhi1998

lancekrogers, 2020-08-20:
What did you have to wget?

goodloop, 2021-01-20:
same issue happened .  I fix out that the download cache for toolchain is not  complete. so uncompress failed.
But setup.sh says it success. it leads to the confusion.

ZeddBao, 2023-09-13:
I met the same issue. The error is caused by the missing toolchain in `UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64` as mentioned above. The toolchain should automatically be downloaded and set up in Setup.sh. The installation failure is likely due to the unsuccessful access to `http://cdn.unrealengine.com/xxxxxx`. You should replace all `http` for `https`. That may fix the problem.