microsoft/AirSim #2832: Record images using API instead of Button

sourabhlal, 2020-07-07:
I am working with a script that runs several simulations with AirSim, and I would like to automate the clicking of the Record button. Is there anyway I could do this using the Python API?

rajat2004, 2020-07-07:
Oh, so you're launching the simulation from scripts themselves? You could use the APIs to fetch the data and store it in the desired format, but if the Recording option gives what is required, then makes sense to use that directly(also might be faster)

There isn't any such API right now, but should be starighforward to add it, you can have a look at https://microsoft.github.io/AirSim/apis/#adding-new-apis-to-airsim which links an example commit. Or might give it a shot myself

rajat2004, 2020-07-07:
I've opened #2834 which adds the APIs

jeffreyng99, 2020-07-17:
@sourabhlal For you scripts, do you run them after you start the simulation by pressing play? I'm trying to figure out how to start the simulation without having to do it manually. I'm working in Linux which may be different than Windows.

rajat2004, 2020-07-18:
@jeffreyng99 If you have set `SimMode`, then AirSim should start without the prompt to select the vehicle. If you're talking about running in the Editor, then yeah, you'll have to press Play in that.

PR #2834 (2020-07-21): Add Recording APIs
Closes #2832 
Contains non-implemented Unity methods, Python docstrings and adds info in `apis.md` as well
closed 2020-07-21 by madratman