microsoft/AirSim #2836: How to get IMU calibration parameters?

kharyal, 2020-07-07:
I am working on Visual-Inertial Odometry, and want to use Airsim to create a synthetic environment to work on. For this, I need IMU calibration parameters (biases and variances) similar to [this](https://github.com/AaltoVision/ADVIO/tree/master/calibration). How can I get the calibration parameters for the IMU?

TIA

rajat2004, 2020-07-08:
Maybe https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/imu/ImuSimpleParams.hpp#L18?

kharyal, 2020-07-09:
> Maybe https://github.com/microsoft/AirSim/blob/master/AirLib/include/sensors/imu/ImuSimpleParams.hpp#L18?

Are these the default values?

kharyal, 2020-07-13:
In the same code, I can find these lines commented.
```
/* ref: Parameter values are for MPU 6000 IMU from InvenSense 
    Design and Characterization of a Low Cost MEMS IMU Cluster for Precision Navigation
    Daniel R. Greenheck, 2009, sec 2.2, pp 17
    http://epublications.marquette.edu/cgi/viewcontent.cgi?article=1326&context=theses_open
    Datasheet:
    https://www.invensense.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf
    For Allan Variance/Deviation plots see http://www.invensense.com/wp-content/uploads/2015/02/MPU-3300-Datasheet.pdf
*/
```
And in the http://epublications.marquette.edu/cgi/viewcontent.cgi?article=1326&context=theses_open link mentioned in the comment, the white noise std for MPU-6000 is mentioned. I can't find the white noise std in the code (I can find only the random walk std).

Thanks