microsoft/AirSim #2838: Multi-Vehicle Subwindows

mhl787156, 2020-07-08:
Hi, I am wanting to live display a camera from each of the drones I have created. Currently in the settings.json, we add the subwindows like so: 
> "SubWindows": [
>    {"WindowID": 0, "ImageType": 0, "CameraName": "3", "Visible": true},
>  {"WindowID": 1, "ImageType": 3, "CameraName": "0", "Visible": true},
>   {"WindowID": 2, "ImageType": 6, "CameraName": "4", "Visible": true}
>  ]

Am I correct in saying that CameraName can only refer to cameras of the 'first' drone? Is there a way to an updated way to display other drones's cameras in the subwindows?

As far as I can find, there have only been two previous issues on this problem in particular: #1172 and #133. #1172 presents a possible solution that someone else has implemented, but their changes no longer work as the api has changed. 

Pulled from master branch about a week ago. Perhaps adding a 'VehicleName' parameter to the Subwindow entries might allow this feature? 

Many Thanks!

rajat2004, 2020-07-08:
I've opened #2841 which adds `Vehicle` option to the Subwindows settings

mhl787156, 2020-07-08:
Amazing thank you!

PR #2841 (2020-09-03): Add Vehicle option for Subwindow settings
Example Settings - 
```
{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
    "SettingsVersion": 1.2,
    "SimMode": "Car",

    "Vehicles": {
        "Car1": {
          "VehicleType": "PhysXCar",
          "AutoCreate": true
        },
        "Car2": {
          "VehicleType": "PhysXCar",
          "AutoCreate": true,
          "X": 0, "Y": 10, "Z": 0
        }
    },
    "SubWindows": [
        {"WindowID": 0, "CameraName": "0", "ImageType": 3, "VehicleName": "Car1", "Visible": false},
        {"WindowID":  [... cut]
closed 2020-09-03 by madratman