microsoft/AirSim #2840: Problem on launching UE4 in a remote server (vnc)

pengzhi1998, 2020-07-08:
Hi everyone.
When I run ./UEeditor, the following error is reported:
```
[2020.07.08-14.28.27:681][  0]LogLinux: Warning: MessageBox: Cannot find a compatible Vulkan device or driver. Try updating your video driver to a more recent version and make sure your video card supports Vulkan.
```
However, I installed Vulkan as this [link](https://linuxconfig.org/install-and-test-vulkan-on-linux) shows. And there is no error reported during installation. But when I `vulkaninfo`, the following error appears: 
```
error: XDG_RUNTIME_DIR not set in the environment.
/build/vulkan-tools-136mCR/vulkan-tools-1.1.126.0+dfsg1/vulkaninfo/vulkaninfo.h:399: failed with ERROR_INITIALIZATION_FAILED
```
Is this the reason why the UE4 cannot be launched? If not, how to solve it?
My system is Ubuntu 18.04 LTS, and Nvidia driver is 430. I'm using UE4.24. (Actually, I have tried OpenGL3 and OpenGL4 as well, both doesn't work)
Thank you so much for your help!!!

kminoda, 2020-07-11:
I also encountered this issue when running UE4.24 on Linux via chrome remote desktop or teamviewer, even though it works pretty well when I launch UE4.24 locally on the same machine.
I still cannot find any solution to this, and ended up using remote windows PC or local Linux PC.

madratman, 2020-07-29:
Are you running this inside a docker container? If so, make sure you run docker with the gpu param (`docker --gpus all / --gpu 1 ...` see https://github.com/NVIDIA/nvidia-docker/tree/master#usage)

Does following https://askubuntu.com/questions/456689/error-xdg-runtime-dir-not-set-in-the-environment-when-attempting-to-run-naut (and https://askubuntu.com/questions/872792/what-is-xdg-runtime-dir) help?

m1baldwin, 2020-07-29:
I have had some success running unreal engine on a google cloud platform VM. In my setup, I used OpenGL4, Virtual GL, and turbovnc. 

What VNC software are you using? I would recommend using a VNC software designed specifically to work with 3d workloads + virtual GL. For example, turboVNC + virtual GL might be an interesting direction for you. I have seen similar approaches in some other work around creating remote-desktop based simulation workstations that run things like gazebo or unity. 

One example is the openuav playground: https://github.com/Open-UAV/openuav-turbovnc

m1baldwin, 2020-07-29:
Also, have you tried running unreal editor just using opengl4, by passing the -opengl4 flag?

huyaoyu, 2021-06-03:
I figured out [a way to run UE4.26 packaged project in a Docker container with Vulkan](https://github.com/microsoft/AirSim/issues/3703#issuecomment-854221381). Maybe related to this issue.