microsoft/AirSim #2842: How to access the API within the UE4 C++ files (not externally)

aaronalmeida, 2020-07-08:
As the title states, I want to be able to use the API's to control a drone within Unreal Engine rather than having another Visual Studio solution running. For example, when I boot my project in Unreal, I want the API's to already start running in the background.  Someone attempted to do it here #1759 and I tried replicating it, but ultimately couldn't get it to work on my own desktop. 

Thank you!

aaronalmeida, 2020-07-30:
Hey,

Thanks for the reply. I meant opening the UE project in C++ and coding the functionality of the AirSim API (as in the moveToPosition(), takeOf() functions) within that UE project file (I am using C++ rather than python). Currently, you would have to have the API code open on one screen while running UE on the other. 

Appreciate any help you could provide me with!

jonyMarino, 2020-12-01:
Hi @aaronalmeida,
Why do you want to do that?