microsoft/AirSim #2846: How to get simulation timestamp for odometry in ROS API when clockspeed<1?

kminoda, 2020-07-11:
I am currently using ROS API built in this repository to get IMU and odometry data for drones in AirSim. In order to get high frequency with a clean timestamp, now I am setting clockspeed=0.1 in`settings.json`.

The issue here is that the way of msg timestamping for IMU and odometry is different. 

IMU uses unreal_timestamp (=simulation time) as in here (from `ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp` at line 654)
```
sensor_msgs::Imu AirsimROSWrapper::get_imu_msg_from_airsim(const msr::airlib::ImuBase::Output& imu_data)
{
    sensor_msgs::Imu imu_msg;
    // imu_msg.header.frame_id = "/airsim/odom_local_ned";// todo multiple drones
    imu_msg.orientation.x = imu_data.orientation.x();
    imu_msg.orientation.y = imu_data.orientation.y();
    imu_msg.orientation.z = imu_data.orientation.z();
    imu_msg.orientation.w = imu_data.orientation.w();

    // todo radians per second
    imu_msg.angular_velocity.x = imu_data.angular_velocity.x();
    imu_msg.angular_velocity.y = imu_data.angular_velocity.y();
    imu_msg.angular_velocity.z = imu_data.angular_velocity.z();

    // meters/s2^m 
    imu_msg.linear_acceleration.x = imu_data.linear_acceleration.x();
    imu_msg.linear_acceleration.y = imu_data.linear_acceleration.y();
    imu_msg.linear_acceleration.z = imu_data.linear_acceleration.z();

    imu_msg.header.stamp = make_ts(imu_data.time_stamp);
    // imu_msg.orientation_covariance = ;
    // imu_msg.angular_velocity_covariance = ;
    // imu_msg.linear_acceleration_covariance = ;

    return imu_msg;
}
```

while odometry (and GPS) msgs use `ros::Time::now()` (=wall time, not simulation time) as below (from `ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp` at line 745)
```
            ros::Time curr_ros_time = ros::Time::now();

            // convert airsim drone state to ROS msgs
            multirotor_ros.curr_odom_ned = get_odom_msg_from_airsim_state(multirotor_ros.curr_drone_state);
            multirotor_ros.curr_odom_ned.header.frame_id = multirotor_ros.vehicle_name;
            multirotor_ros.curr_odom_ned.child_frame_id = multirotor_ros.odom_frame_id;
            multirotor_ros.curr_odom_ned.header.stamp = curr_ros_time;
```

Are there any solution to get simulation timestamp for odometry as well as IMU?

GimpelZhang, 2020-07-13:
I think you can also use the "make_ts" function for the timestamps of odometry and GPS messages, just like imu_msgs does.

kminoda, 2020-07-13:
Thanks for the comment! Actually what I meant here was that I couldn’t figure out how to get unreal_time for odom/gps, which is easy for IMU msg. If I can get this, as you mentioned, I can just use “make_ts” function to generate ROS timestamp.
But I think I reached a solution. I will post it if it works well.

kminoda, 2020-07-14:
I could solve this problem by changing line 746 in `ros/src/airsim_ros_pkgs/src/airsim_ros_wrapper.cpp` as below
```
row::Time curr_ros_time = make_ts(multirotor_ros.curr_drone_state.timestamp);
```

closed 2020-07-14 by kminoda