microsoft/AirSim #2848: Deep Q learning

johan606303, 2020-07-11:
Hi.

I am developping a DQL for drones to teach how to avoid obstacles. I saw some tutorials, there they use space as input state. I'm want my drone to learn avoiding obstacles with using camera (ImageType.DepthPerspective). What do I have to use as input state for my DQL algorithm ?
Thank you for help.

aqeelanwar, 2020-07-16:
You can check this repository based on AirSim
https://github.com/aqeelanwar/PEDRA

madratman, 2020-07-29:
"What do I have to use as input state for my DQL algorithm ?" 
there's no one single correct answer to this question. 
A classical navigation pipelines first does state estimation using visual + inertial sensor data, uses that to build a map, and then uses both the state estimation and an inferred map of the environment to do motion planning level tasks. 
Now, if you want to do deep RL to do the above, first you have to define and scope the extent of your (research) work, and make assumptions. 
For example, you may assume a perfect state estimate and a perfect sensor model, and then focus only on the navigation aspect of the task. In that case, your inputs would be the state estimate and the sensor data, and possibly the goal itself (for a goal-conditioned policy). 
Or you can be a bit more ambitious, and try to do everything with only vision sensor data as input (learning to navigate without a state estimate, for example). 
Or you make your life tougher by making your project more realistic and choose to operate on noisy inertial sensors and vision sensor data as input.

closed 2020-07-29 by madratman