microsoft/AirSim #2850: How to make a connection between ROS on ubuntu and Airsim in another machine (windows)

lucafei, 2020-07-14:
i run the ubuntu 16.04 and ros with version of kinetic. Moreover the airsim is running on the windows10.
Now i want to make a communication between ros and airsim so that i make some trajectory planning, perception, model training and so on in ros. Does it have some methods to realize this?

mhl787156, 2020-07-15:
Hey, I've been trying to do this just today, I think you can specify host_ip of Airsim on Windows by setting a parameter when you launch an airsim client node: See line 12 in [source](https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/src/airsim_node.cpp), let me know if this works for you?

I've been trying to use ardupilot, mavlink and hopefully mavros but I havn't quite got it working yet.

rajat2004, 2020-07-16:
You can try changing `localhost` to the IP address of the Windows machine here - https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_node.launch#L2
The PR which added this option - https://github.com/microsoft/AirSim/pull/2368
The docs need to be updated with the info, if someone does try it out, then a PR for improvements to the docs will be great!

@mhl787156 For Using AP+AirSim+ROS, you could try using the scripts mentioned here - https://ardupilot.org/dev/docs/sitl-with-airsim.html#ros-with-multi-vehicle-simulation. It's described for multiple vehicles but should work with changing it to a single vehicle

lucafei, 2020-07-16:
> Hey, I've been trying to do this just today, I think you can specify host_ip of Airsim on Windows by setting a parameter when you launch an airsim client node: See line 12 in [...]

Hi, mhl787156, 
firstly thanks for your advice, i have tried this api. I think it has already worked. After you installed the ros_wrapper for airsim, you can just change the host_ip as ip address of your local machine, like @rajat2004 said. By now i'm just tesing the connection between ros and airsim, so i can't say anything about usage of provided api.

lucafei, 2020-07-16:
> You can try changing `localhost` to the IP address of the Windows machine here - https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_node.launch#L2
> The PR which added this option - #2368
> The docs need to be updated with the info, if someone does try it out, then a PR for improvements to the docs will be great!
> 
> @mhl787156 For Using AP+AirSim+ROS, you could try using the scripts mentioned here - https://ardupilot.org/dev/docs/sitl-with-airsim.html#ros-with-multi-vehicle-simulation. It's described for multiple vehicles but should work with changing it to a single vehicle
> There's also the related Discuss topic which might be helpful - https://discuss.ardupilot.org/t/airsim-multiple-arducopter-sitl-mavros/49613

hi rajat2004,

you're right. i have tried this method. the connection works.

kminoda, 2020-07-20:
FYI, I am currently using AirSim + ROS only on windows 10 by using WSL (ubuntu 18.04 + ROS melodic). Though I had some problems with GUI (such as rviz and rqt_image_view), now it's working fine.

lucafei, 2020-07-20:
> FYI, I am currently using AirSim + ROS only on windows 10 by using WSL (ubuntu 18.04 + ROS melodic). Though I had some problems with GUI (such as rviz and rqt_image_view), now it's working fine.

hi, kminoda, maybe you can try the python api of airsim, if you don't have to use ros. by the way did you have tried the domain randonmization of airsim?

kminoda, 2020-07-20:
@lucafei No. Since I am new to AirSim&UE4, didn't even know that you can do this kind of thing easily in AirSim.

kartikeya13, 2020-07-21:
Hii @lucafei,
It might sound silly, but I am a newbie to ROS and I want to connect AirSim (windows) with ROS ubantu for extracting camera images from AirSIm simulator. If you could just let me know how you did that in a layman way I would be so thankful to you.

lucafei, 2020-07-22:
> Hii @lucafei,
> It might sound silly, but I am a newbie to ROS and I want to connect AirSim (windows) with ROS ubantu for extracting camera images from AirSIm simulator. If you could just let me know how you did that in a layman way I would be so thankful to you.

you can find the api for getting image in https://microsoft.github.io/AirSim/image_apis/

kartikeya13, 2020-07-22:
@lucafei 
Can you let me know how I can get frames using that api and process it through a loop to get a video displayed in an opencv window?

madratman, 2020-07-29:
@kartikeya13 https://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_gui/py_image_display/py_image_display.html

closed 2020-07-29 by madratman

Yueting-Li, 2020-10-04:
Hey, i am currently using Airsim on the Windows10, and i want to run px4 + ros + mavros on another machine(ubuntu). So I use a cable to connect the two computers. I set Windows ip address as 192.168.1.1, and ubuntu ip as 192.168.1.2. Here is my airsim settings.json:
{
     "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
     "SettingsVersion": 1.2,
     "SimMode": "Multirotor",
  "Vehicles": {
    "mav": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "LocalHostIp": "192.168.1.1",
      "SitlIp": "192.168.1.2",
      "SitlPort": 14556,
      "UdpIp": "192.168.1.2",
      "UdpPort": 14560,
      "QgcHostIp": "192.168.1.2",
      "QgcPort": 14550,
      "Sensors": {
      "LidarCustom": {
        "SensorType": 6,
        "Enabled": true,
        "NumberOfChannels": 16,
        "PointsPerSecond": 10000,
        "X": 0,
        "Y": 0,
        "Z": -1,
        "DrawDebugPoints": true
      }
    },
    "Cameras": {
      "front_center_custom": {
        "CaptureSettings": [
          {
            "PublishToRos": 1,
            "ImageType": 0,
            "Width": 672,
            "Height": 376,
            "FOV_Degrees": 90,
            "TargetGamma": 1.5
          },
          {
            "PublishToRos": 1,
            "ImageType": 2,
            "Width": 672,
            "Height": 376,
            "FOV_Degrees": 90,
            "TargetGamma": 1.5
          }
        ],
        "X": 0.50,
        "Y": 0,
        "Z": 0.10,
        "Pitch": 0.0,
        "Roll": 0.0,
        "Yaw": 0.0
      }
    },
    "X": 2,
    "Y": 0,
    "Z": 0,
    "Pitch": 0,
    "Roll": 0,
    "Yaw": 0
   }
  }
  }

Then I successfully run PX4 on ubuntu machine and connected to airsim. But when I tried changing localhost (in airsim_node.launch) to the IP address 192.168.1.1. like @rajat2004 said.  It doesn't work. The terminal always show connecting××××××××××× after i runned airsim_node.launch on ubuntu. Can anybody give me some suggestions? 
Maybe I did something wrong. Thanks a million!

Yueting-Li, 2020-10-06:
Hey, i am currently using Airsim on the Windows10, and i want to run px4 + ros + mavros on another machine(ubuntu). So I use a cable to connect the two computers. I set Windows ip address as 192.168.1.1, and ubuntu ip as 192.168.1.2.

Then I successfully run PX4 and mavros on ubuntu machine and connected to Airsim. But when I tried to bring Airsim ROS nodes into the picture, it doesn't work.
I just changed "host_ip"(in airsim_node.launch) to windows IP address 192.168.1.1.
Then I runned Airsim on windows and airsim_node.launch on ubuntu. The terminal showed "Waiting for connection - XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"
Can anybody give me some suggestions? Thanks a million!

jpapon, 2020-10-15:
If anyone comes across this trying to get ROS working in WSL, you'll need to set the host machine IP here:
```
https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_node.launch#L2
```
This is the actual IP address of you Windows machine - similar to how you set up X forwarding, you can't use localhost.

This worked for me - before I did it, I could see topic names, but no data came out of Unreal.