microsoft/AirSim #2851: Inherent y drift with increasing clockspeed on multirotor

amaalo2, 2020-07-14:
Hello, 

When commending a UAV to go in a straight line, the UAV drifts more to the left as the clockspeed increases. I use the moveByVelocityZAsync API. The columns shown below are x pos, y pos and z pos. I use Python 3.7, and the master version of Airsim corresponding to commit hash 827f1f55df68eb1d026390e6db06c9dd11152ba2 (The Master without the Azure update) as well as Windows 10, and this the rig I am using. 
[image]

I am running the experiment on my own custom environment where the drone just takes off and goes straight. If I cannot match real-time performance, let's say ~0.1 m drift on y then there is no guarantee that the trained agent on the faster than real-time environment will able to reproduce similar results in the real-time environment.

My custom environment
[image]

Real-Time results

[image]

15x Real-time results
[image]

100x Real-Time results

[image]

Here is the python script I used to run the test. 

` import setup_path
import airsim
import pprint
import time

for i in range(10):

    client = airsim.MultirotorClient()
    client.reset()
    client.confirmConnection()
    client.enableApiControl(True)
    client.armDisarm(True)
    client.takeoffAsync().join()

   
    while True:
        collision_info = client.simGetCollisionInfo()
        client.enableApiControl(True)

        #Move Straight
        client.moveByVelocityZAsync(1, 0, -2, 1).join()

        quad_state = client.simGetGroundTruthKinematics().position
        pos_x = quad_state.x_val
        pos_y = quad_state.y_val
        pos_z = quad_state.z_val

        print(pos_x,pos_y,pos_z)

        if collision_info.has_collided:
            print(i)
            print("Collision at pos %s, normal %s, impact pt %s, penetration %f, name %s, obj id %d" % (
                pprint.pformat(collision_info.position), 
                pprint.pformat(collision_info.normal), 
                pprint.pformat(collision_info.impact_point), 
                collision_info.penetration_depth, collision_info.object_name, collision_info.object_id))               
            break
`

To me this is a high priority as RL training is simply too costly in terms of time to run the experiment in real-time.
Thanks,
Anthony

amaalo2, 2020-07-14:
I was able to reduce the Y drift by playing with the PID on the Y value and adding the following line under 
`client.takeoffAsync().join()`

`client.setVelocityControllerGains(velocity_gains=airsim.VelocityControllerGains(airsim.PIDGains(0.2,0,0),airsim.PIDGains(2.0,0,0),airsim.PIDGains(2,2,0)))`

Here are the results

[image]

However, this is a patch, and not an acceptable solution.

WreckItTim, 2023-01-25:
Any updated solutions to this?

I am running into the same issue. I am running some RL script which issues several commands to AirSim. As I am watching, I can see that occasionally an episode will get some inherent drift with it for the entirety of the episode. After calling reset, it seems to mitigate the issue at least for the next episode.

I see the same behavior when I send commands to the multirotor client one at a time. Occasionally, after calling moveByVelocityAsync().join(), the drone will start to spin around the vertical axis with fixed yaw. It's like it is trying to stabilize itself to stay still. However, the spins can get very erratic. This still happens even at a clock speed of 1.

I think it is related to hover mode, but can't prove it.

amaalo2, 2023-01-26:
Hi Tim, 

I did not end up solving this problem. Instead, I used a different simulator that is more stable and better suited for RL  in my opinion. 

https://github.com/utiasDSL/gym-pybullet-drones/issues

WreckItTim, 2023-01-26:
Thanks for the response. I have been meaning to check out pybullet. AirSim has so many resources available, but with them being depreciated soon... =( 

Hopefully they will still be able to address this issue before closing up shop! I opened a new ticket with some script that can repeat and show the exact problem which I found after some trial and error: [https://github.com/microsoft/AirSim/issues/4780](url).

xxEoD2242, 2023-01-26:
@WreckItTim Hey Tim,

We have decided to continue on AirSim as an open source project with a new name called Colosseum and are working to better support RL. We are in the process of updating this to Unreal Engine 5 and improving the underlying architecture so that it can support more use cases.

You can find it here: [Colosseum](https://github.com/CodexLabsLLC/Colosseum.git)

xxEoD2242, 2023-01-26:
We also have a slack where you can ask questions. I'm the owner of the company that is supporting it and I also work with it daily so happy to help. [Colosseum Slack](https://join.slack.com/t/colosseum-sim/shared_invite/zt-1nx1worza-XNsS_KkXuiKvbm~e2E8Big)

From my experience, I don't really use their Velocity or Position controllers, as they try to normalize these values to get a consistent speed, which isn't what I needed.

WreckItTim, 2023-01-26:
That's awesome! I will look into using Colosseum for future deployments. Currently I have a paper submission deadline I am trying to make, so had to run with my current deployment and use a stop-gap to handle the drift in AirSim. If you are continuing working on Colosseum from an AirSim fork, could you please see the ticket I opened about the y-axis drift: https://github.com/microsoft/AirSim/issues/4780

There is a link to a YouTube video showing the exact problem. I was able to isolate it, and it can potentially be a problem for Colosseum. Thanks!

What do you use to control the drone from your RL environment, if not the velocity and position controllers/commands? I typically use moveByVelocityAsync().join()