microsoft/AirSim #2856: Mapping Unreal/Airsim Environment using ROS packages

thejeshk, 2020-07-15:
Hello Guys, 

Is there any possible way to map the custom Unreal/Airsim environment using ROS packages? 

Thank you :)
Thejesh

manthan99, 2020-07-16:
ORB SLAM works fine with Airsim if you want to create a feature-based map using monocular camera data. The calibration parameters are pretty straight forward. Since you already have the ground truth odometry, you may try the common ROS mapping packages.

Yueting-Li, 2020-10-14:
Hi,
I am running px4 sitl. And I want to use ORB SLAM to map Unreal envionment. I start airsim_ros_wrapper node to get images from unreal envionment, then I use xbox to control drone but unfortunately the uav suffers great instability. If I close airsim node, the drone can return to stable state. Can anybody give me some suggestions?
Thank you!

manthan99, 2020-10-14:
Hi @Yueting-Li 
Try changing the flight mode to position control mode using the QGround Control App. It will be much more stable than the default stabilized  flight mode

Yueting-Li, 2020-10-14:
Thank you! @manthan99 
Actually I have changed to position control mode. If I just run px4,  everything is ok. But if I run airsim_node.launch in the meantime, the drone would fly without control. It confused me for a long while.

manthan99, 2020-10-14:
@Yueting-Li , What settings.json are you using? Have you changed the settings according to PX4 instead of simple Flight controller?

Yueting-Li, 2020-10-14:
@manthan99
My settings.json is as follows:
{
     "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
     "SettingsVersion": 1.2,
     "SimMode": "Multirotor",
  "Vehicles": {
    "mav": {
      "VehicleType": "PX4Multirotor",
      "UseSerial": false,
      "LocalHostIp": "192.168.1.1",
      "SitlIp": "192.168.1.2",
      "SitlPort": 14556,
      "UdpIp": "192.168.1.2",
      "UdpPort": 14560,
      "QgcHostIp": "192.168.1.2",
      "QgcPort": 14550,
      "params": {
          "NAV_RCL_ACT": 0,
          "NAV_DLL_ACT": 0,
          "LPE_LAT": 47.641468,
          "LPE_LON": -122.140165,
          "COM_OBL_ACT": 1
      },
      "Sensors": {
      "Barometer": {
        "SensorType": 1,
        "Enabled": true
      },
      "Imu": {
        "SensorType": 2,
        "Enabled": true
      },
      "Gps": {
        "SensorType": 3,
        "Enabled": true
      },
      "Magnetometer": {
        "SensorType": 4,
        "Enabled": true
      },
      "Distance": {
        "SensorType": 5,
        "Enabled": true
      },
      "LidarCustom": {
        "SensorType": 6,
        "Enabled": true,
        "NumberOfChannels": 16,
        "PointsPerSecond": 10000,
        "X": 0,
        "Y": 0,
        "Z": -1,
        "DrawDebugPoints": true
      }
    },
    "Cameras": {
      "front_center_custom": {
        "CaptureSettings": [
          {
            "PublishToRos": 1,
            "ImageType": 0,
            "Width": 672,
            "Height": 376,
            "FOV_Degrees": 90,
            "TargetGamma": 1.5
          },
          {
            "PublishToRos": 1,
            "ImageType": 2,
            "Width": 672,
            "Height": 376,
            "FOV_Degrees": 90,
            "TargetGamma": 1.5
          }
        ],
        "X": 0.50,
        "Y": 0,
        "Z": 0.10,
        "Pitch": 0.0,
        "Roll": 0.0,
        "Yaw": 0.0
      }
    },
    "X": 2,
    "Y": 0,
    "Z": 0,
    "Pitch": 0,
    "Roll": 0,
    "Yaw": 0
   }
  },

    "SubWindows": [
       {"WindowID": 0, "ImageType": 0, "CameraName": "front_center_custom", "Visible": true},
       {"WindowID": 1, "ImageType": 3, "CameraName": "front_center_custom", "Visible": true},
       {"WindowID": 2, "ImageType": 5, "CameraName": "front_center_custom", "Visible": true}
    ]
  }

PS: "192.168.1.1" is the ip address where I run airsim&UE4(on windows),  "192.168.1.2" is the ip address where I run PX4 & ROS(on ubuntu).

Yueting-Li, 2020-10-14:
@manthan99,
My settings.json is as follows:
{
"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
"SettingsVersion": 1.2,
"SimMode": "Multirotor",
"Vehicles": {
"mav": {
"VehicleType": "PX4Multirotor",
"UseSerial": false,
"LocalHostIp": "192.168.1.1",
"SitlIp": "192.168.1.2",
"SitlPort": 14556,
"UdpIp": "192.168.1.2",
"UdpPort": 14560,
"QgcHostIp": "192.168.1.2",
"QgcPort": 14550,
"params": {
"NAV_RCL_ACT": 0,
"NAV_DLL_ACT": 0,
"LPE_LAT": 47.641468,
"LPE_LON": -122.140165,
"COM_OBL_ACT": 1
},
"Sensors": {
"Barometer": {
"SensorType": 1,
"Enabled": true
},
"Imu": {
"SensorType": 2,
"Enabled": true
},
"Gps": {
"SensorType": 3,
"Enabled": true
},
"Magnetometer": {
"SensorType": 4,
"Enabled": true
},
"Distance": {
"SensorType": 5,
"Enabled": true
},
"LidarCustom": {
"SensorType": 6,
"Enabled": true,
"NumberOfChannels": 16,
"PointsPerSecond": 10000,
"X": 0,
"Y": 0,
"Z": -1,
"DrawDebugPoints": true
}
},
"Cameras": {
"front_center_custom": {
"CaptureSettings": [
{
"PublishToRos": 1,
"ImageType": 0,
"Width": 672,
"Height": 376,
"FOV_Degrees": 90,
"TargetGamma": 1.5
},
{
"PublishToRos": 1,
"ImageType": 2,
"Width": 672,
"Height": 376,
"FOV_Degrees": 90,
"TargetGamma": 1.5
}
],
"X": 0.50,
"Y": 0,
"Z": 0.10,
"Pitch": 0.0,
"Roll": 0.0,
"Yaw": 0.0
}
},
"X": 2,
"Y": 0,
"Z": 0,
"Pitch": 0,
"Roll": 0,
"Yaw": 0
}
},
"SubWindows": [
   {"WindowID": 0, "ImageType": 0, "CameraName": "front_center_custom", "Visible": true},
   {"WindowID": 1, "ImageType": 3, "CameraName": "front_center_custom", "Visible": true},
   {"WindowID": 2, "ImageType": 5, "CameraName": "front_center_custom", "Visible": true}
]
}

PS: "192.168.1.1" is the ip address where I run airsim&UE4(on windows), "192.168.1.2" is the ip address where I run PX4 & ROS(on ubuntu).

manthan99, 2020-10-16:
@Yueting-Li Sorry for the late reply. Seems fine. I was using almost similar settings and was getting a quite stable flight in the position-controlled mode. I assume that you are using the "airsim_node.launch" file and not the other launch files (since some of them are for controlling the drone using ros commands)
Let me know if you find a solution

NickPerezCarletonUniversity, 2021-07-08:
does anyone in this thread have experience running active SLAM in airsim using ROS? as in, using a SLAM algorithm to decide what the next control input is for flying the drone based off the current state estimate?

xxEoD2242, 2021-07-08:
@NickPerezCarletonUniversity that is what we are currently doing. It is possible but there is a lot you will need to do that is not in the guide. Can discuss offline if you'd like.

NickPerezCarletonUniversity, 2021-07-08:
that would be awesome. send me a message on linkedin?... https://www.linkedin.com/in/nicolas-perez-is-awesome/

Erickrk, 2022-02-25:
Hello everyone, any updates on this?