microsoft/AirSim #2857: AttributeError: module 'msgpack' has no attribute 'Packer from airsim?

yash1315, 2020-07-16:
I war trying to implement reinforcement learning algorithms and was working on it for about 4-5 days. But today I don't what happened, the airsim package has some issue. The error is shown below. 

Traceback (most recent call last):
  File "pg2.py", line 5, in <module>
    import airsim
  File "C:\Users\lenovo\Anaconda3\lib\site-packages\airsim\__init__.py", line 1, in <module>
    from .client import *
  File "C:\Users\lenovo\Anaconda3\lib\site-packages\airsim\client.py", line 3, in <module>
    from .utils import *
  File "C:\Users\lenovo\Anaconda3\lib\site-packages\airsim\utils.py", line 10, in <module>
    from .types import *
  File "C:\Users\lenovo\Anaconda3\lib\site-packages\airsim\types.py", line 2, in <module>
    import msgpackrpc #install as admin: pip install msgpack-rpc-python
  File "C:\Users\lenovo\Anaconda3\lib\site-packages\msgpackrpc\__init__.py", line 6, in <module>
    from msgpackrpc.server import Server
  File "C:\Users\lenovo\Anaconda3\lib\site-packages\msgpackrpc\server.py", line 81, in <module>
    class _Responder:
  File "C:\Users\lenovo\Anaconda3\lib\site-packages\msgpackrpc\server.py", line 87, in _Responder
    def set_result(self, value, error=None, packer=msgpack.Packer()):
AttributeError: module 'msgpack' has no attribute 'Packer'

Can anyone help me with this issue? 
Thanks

Abc11c, 2020-07-29:
I had a similar issue try downgrading `tornado=4.5.3` in Anaconda the latest version seems to have conflict with airsim `pip install airsim` module.
Hope it helps

LiorRaines, 2020-10-08:
I had the same problem, installing msgpack-python solved it for me

jonyMarino, 2021-03-22:
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

closed 2021-03-22 by jonyMarino

BetterLL, 2023-07-29:
I have the same question, how did you resolve it?