microsoft/AirSim #2859: Domain randomization for airsim

lucafei, 2020-07-17:
my situation: airsim on windows pc, ros wrapper for airsim on ubuntu16.04 pc, unreal engine 4.21
### Question:
i want to make domain randomization to train my model in airsim so that i can transfer my trained model in real hardware. I have seen some infos about domain randomization in airsim tutorial (https://microsoft.github.io/AirSim/retexturing/) but it has only the function to change the texture. If i want to random more factors like lightning, shape, and so on, how can i get this? Does airsim support these? 
Thank you very much in advance.

madratman, 2020-07-29:
We only support changing of textures. 
cc @saihv

lucafei, 2020-07-29:
> We only support changing of textures.
> cc @saihv

Hello, madratman, thank you for your answer. but does it have any possible ways to realize more factors randomization? it is possible to realize the purpose in airsim by my own  ? Can you give me some guidance for that? Thank you very much in advance.

Fraser-Paine, 2020-08-03:
Hi @lucafei, I'm not certain that this is what you're looking for but there are a few other methods that may help you with domain randomisation that you can find [here](https://github.com/microsoft/AirSim/blob/fd4a8fecfc873259848293e4019b3ad31aee66c6/PythonClient/airsim/client.py#L318).

The functions needed to randomise position and scale are simGetObjectPose, simSetObjectPose, simGetObjectScale and simSetObjectScale.
This plus texture swapping should get most of what you're looking for.

Unfortunately setting poses for rigged models doesn't seem to be supported since the Character API was removed a few months ago.