microsoft/AirSim #2863: Ray-cast to Get Velocity of Moving Multicopters Only Return 0

dnychennnn, 2020-07-20:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

dnychennnn, 2020-07-20:
Hi,

I've tried to mimic CARLA's implementation for radar in Airsim. When it came to compute the response of doppler velocity, I have the following implementation:

[image]

And then the question came across that the returned velocity value of hit actor(in my case it's another drone agent) by the Unreal ray cast is always 0. I validated this by hovering two copters 20m away at the same height and flying the target towards the observer(ray cast source), as shown below for left being the target and right being the radar source,

[image]

What could be the problem that the actor velocity only returns 0?

rajat2004, 2020-07-20:
Looks strange, are the actor name and the location correct?
One reason I could think of, based on [`GetVelocity()` doc](https://docs.unrealengine.com/en-US/API/Runtime/Engine/GameFramework/AActor/GetVelocity/index.html) could be that Airsim is using its own physics engine for multirotors, which might cause velocity to be shown as zero. Though this is a pretty far-fetched thought, and need to check the asset.
Maybe try using this with Car since it uses PhysX and see if there's any change?

Radar sensor will be a great addition!

dnychennnn, 2020-07-21:
@rajat2004 As you can see in the figure below, the position and actor name are fine(maybe a bit of deviation), so I really can't think of a reason why the velocity won't work.
[image]

madratman, 2020-07-29:
Hmm, pretty interesting issue. Yes, the radar implementation would be a great PR. 
Now, in the multirotor mode, we essentially directly set the poses of the multirotor pawn object to be what airsim's custom physics engines tells us. I think this is the reason for unreal pawn to have zero velocity (so for multirotor, the only way to get true velocity is to grab it from multirotor's physics body, and not via an unreal engine api)

In general. the following lines define the "Unreal Engine thread" of AirSim 
https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldBase.cpp#L107-#L126

Depending on the simmode, the call to `updateRendering` will be different.
In the Multirotor case, we simply set the pose of the unreal engine pawn, in the L111 below:

https://github.com/microsoft/AirSim/blob/1f1188a2974b2fe8dafcfc01377bd758b0457246/Unreal/Plugins/AirSim/Source/Vehicles/Multirotor/MultirotorPawnSimApi.cpp#L90-#L112

To get the correct velocity if and when radar hits the multirotor pawn, you'd have to do an if-else on the hittedactor and deduce if it's of the multirotor kind, then you'd have to grab the velocity from the physics body associated with the actor. This association seems a bit tricky to handle, and might warrant some refactoring in the radar impl