microsoft/AirSim #2864: could not find function getMultirotorstate

lucafei, 2020-07-20:
i have two computer, one runs ros on Ubuntu 16.04, another one runs windows10 with airsim simulator. I have connected the ros with airsim on windows, which can be achieved with usage of roswrapper of airsim. But when I use client.getMultirotorstate to get states of drone from airsim. It turned out error: the Programm can’t find the function getMultirotorstate. Does anyone have met same problem? How do you solve this? Thank you in advance.

EdwinMeriaux, 2020-07-20:
can you show the code you are using that has this issue?

lucafei, 2020-07-22:
> can you show the code you are using that has this issue?

hi, EdwinMeriaux, i just run the demo code from airsim_ros_pkags. you can find it in https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_node.launch

EdwinMeriaux, 2020-07-23:
> > can you show the code you are using that has this issue?
> 
> hi, EdwinMeriaux, i just run the demo code from airsim_ros_pkags. you can find it in https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_ros_pkgs/launch/airsim_node.launch

wait so the only thing you are launching is the airsim_ros_pkgs? Are you launching any code on top of that?
Also on a side note. If you do not launch airsim_ros_pkgs have you tried the hello_drone.py code? does that work properly?
If yes try running the getMulitororstate.

so something like:
pos = client.getMultirotorState().kinematics_estimated.position inside of the hellodrone code

madratman, 2020-07-29:
The above comment answered the question. 
If not, can you share a python snippet. 
getMultirotorState is a python API, and has nothing to do with the ros wrapper. 
If you're using the ros wrapper, you need to subscribe to the odometry topic

closed 2020-07-29 by madratman