microsoft/AirSim #2870: Does AirSim support Pixhawk 4 for HITL?

galgyeony, 2020-07-22:
Since Pixhawk 2.1 was discontinued, I have been trying to use Pixhawk 4 for HITL instead.

However, it is unclear whether AirSim supports Pixhawk 4 for HITL since it is not listed as a supported hardware (see the link below):
https://microsoft.github.io/AirSim/px4_setup/

Does AirSim support Pixhawk 4 for HITL? If not, is there another good option for HITL?

Thanks in advance.

galgyeony, 2020-08-13:
I tested with Pixhawk 4 for HITL and it worked successfully.
 
I think it would be better if Pixhawk 4 is listed as a supported hardware in the following link.
https://microsoft.github.io/AirSim/px4_setup/

jonyMarino, 2021-02-02:
Added to the list. Thanks!

closed 2021-02-02 by jonyMarino