microsoft/AirSim #2875: image_data_float always returns an empty image (i.e. list composed of [0.0])

rallen10, 2020-07-22:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

This seems to be a persistent, unaddressed problem based on other issues dating back more than a year (https://github.com/microsoft/AirSim/issues/2089, https://github.com/microsoft/AirSim/issues/2169, https://github.com/microsoft/AirSim/issues/2665, https://github.com/microsoft/AirSim/issues/1840). The only difference for me is that 1. I am running in Unity, and 2. the image is _always_ empty, therefore the workaround to discard image and grab the next one if it is empty doesn't work for me.

To reproduce:
```
import airsim
import numpy as np
client = airsim.MultirotorClient(timeout_value = 7200)
client.confirmConnection()
while True:
        response = client.simGetImages([airsim.ImageRequest("1", airsim.ImageType.DepthPerspective, pixels_as_float=True, compress=False)])[0]
        img1d = np.array(response.image_data_float, dtype=np.float)
        if len(img1d) != response.height * response.width:
            print('empty or bad image. Retrying...')
        else:
            print('properly formatted image!')
            break
```

rallen10, 2020-07-23:
I found pull request https://github.com/microsoft/AirSim/pull/2812 from a similar issu: https://github.com/microsoft/AirSim/issues/2811. I tried merging the pull request locally and [rebuilding AirSim and the Unity wrapper](https://microsoft.github.io/AirSim/Unity/#linux), but to no effect. Still getting the same problem with an empty image

chanho-code, 2020-08-18:
Same problem in Unity

weigangwen, 2020-11-08:
I encountered the same problem in Unity, did you solve it?

SworterW, 2022-11-11:
I also used unity to develop, and found the same problem as yours in the latest airsim version. Have you finally solved it

sandilyasg, 2023-11-06:
@SworterW @weigangwen @chanho-code @rallen10 I am also facing the same issue (using Unity), were you able to solve this issue?

rallen10, 2023-11-07:
Sorry I haven't looked at this work in years. Honestly can't remember if I found a solution, a workaround, or just abandoned this part of my project all together. 

Not very helpful, I know, but figured I shouldn't leave you all hanging

sandilyasg, 2023-11-07:
@rallen10 Thanks for your reply! Are you able to share the github repo of your project?