microsoft/AirSim #2882: Question regarding simple_flight drone dynamics

HelloEgg, 2020-07-27:
Hello. 

I am trying to validate your drone model based on your paper and observed position and orientation value from simulation.

I am trying to see if my input values that goes through re-implemented drone dynamical model matches the observed value.

The paper that I am referencing is "AirSim: High-Fidelity Visual and PhysicalSimulation for Autonomous Vehicles"

(https://arxiv.org/pdf/1705.05065.pdf)

However, I had some problem understanding how to calculate force.

As far as I understood, in page 5, the force generated by propellers are given as follows. 

[image]

However, the variables on the left side are scalar or numeric value while the force seems to be vector (as it is bolded).

How are x,y,z force components calculated?

May I know the reference or part of code that explains this part?

Thank you!

rajat2004, 2020-07-27:
[`set_wrench()`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/RotorActuator.hpp#L109) in `RotorActuator.hpp` and [`RotorParams.hpp`](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/RotorParams.hpp) might be what you're looking for
The direction is the normal of the vertices

HelloEgg, 2020-07-27:
@rajat2004 Thank you very much for your kind reply. 

As I understand, vector Force should be normal*Force*air_density, correct?

According to RotorParams.hpp, it looks like air_density is 1.225f. 

As for normal vector, do I simply get normal vector by converting quaternion to euler angle? 

Thank you so much.

madratman, 2020-07-29:
Hello @HelloEgg
Thrust is (always) generated along the normal vector, as @rajat2004 hinted.  
By default, this normal vector is (0,0,-1) in NED system. Or in other words simply the "up" vector if you don't modify the spawn orientation in settings json. Or in the other words, always along the Z direction in the "body frame" of the quadrotor (where body frame would have some relative orientation w.r.t. the "world frame") 
This obviously will change if the drone is flying around (and has a non-identity orientation), and will correspond to the "Z" direction of the rotation matrix corresponding to the drone's quaternion. 
So, yeah, you can convert the quaternion to a rotation matrix, and the three elements of the third column of the rotation matrix will define the direction along which the thrust is generated.     

Now, let's move from English to C++. 
[setWrench](https://github.com/microsoft/AirSim/blob/master/AirLib/include/vehicles/multirotor/RotorActuator.hpp#L109) indeed converts this from a scalar value to a vector. 
However, this begets the question "if getNormal() is returning (0,0,-1), where is this generate thrust / force converted from body frame to world frame" 
This happens in L291 below 
https://github.com/microsoft/AirSim/blob/4cddf3fb1b4f81083296d5269ea50afb67558381/AirLib/include/physics/FastPhysicsEngine.hpp#L291-L312

Note that getBodyWrench is called by [getNextKinematicsNoCollision](https://github.com/microsoft/AirSim/blob/4cddf3fb1b4f81083296d5269ea50afb67558381/AirLib/include/physics/FastPhysicsEngine.hpp#L314), which is called by [updatePhysics](https://github.com/microsoft/AirSim/blob/4cddf3fb1b4f81083296d5269ea50afb67558381/AirLib/include/physics/FastPhysicsEngine.hpp#L68), which is called by [update()](https://github.com/microsoft/AirSim/blob/4cddf3fb1b4f81083296d5269ea50afb67558381/AirLib/include/physics/FastPhysicsEngine.hpp#L41)

closed 2020-07-29 by madratman

HelloEgg, 2020-07-31:
@madratman  Hi! Thanks for the kind reply. 

I have another question regarding estimating position. 

In L407,

https://github.com/microsoft/AirSim/blob/4cddf3fb1b4f81083296d5269ea50afb67558381/AirLib/include/physics/FastPhysicsEngine.hpp#L403-L407

I noticed that position is only calculated by current velocity and acceleration.

As I understand, according to the code, 

current_pose.position + avg_linear

and avg_linear is calculated as below:

https://github.com/microsoft/AirSim/blob/4cddf3fb1b4f81083296d5269ea50afb67558381/AirLib/include/physics/FastPhysicsEngine.hpp#L345-L347

which is calculated with current acceleration and velocity,
not with anticipated next acceleration and velocity that are driven by Force net.  

This seems to be different from what is written in paper, which calculates next pose from anticipated velocity and acceleration values.

Did I got this correctly? if it is, when is next velocity and acceleration used? 
Is it only used during `getNextKinematicsOnCollision` ?

Thank you