microsoft/AirSim #2884: How to allow multiple drones to communicate each other?

GeneLYJ, 2020-07-27:
I'm pretty new in AirSim, I managed to create multiple drones in the simulation. I just want to make Drone 1 to say hi to Drone 2. The Drone 2 will reply back to Drone 1 once it received hi from Drone 1.

madratman, 2020-07-29:
We don't model multi-agent communication in the C++ backend, however you can do all of that on the client side.
You can use the python APIs to get the state of both drones, and corrupt the state of drone 1 as observed by drone 2 as needed. 
(My assumption here is that by "hi" you refer to something like "state / control inputs of drone i" as seen by "drone j"

closed 2020-07-29 by madratman

GeneLYJ, 2020-07-29:
Yes, if that's the case, may I know which API could read the state of the drone ? I would like to do something like Drone 2 would move to the left when Drone 1 moves to the left.

madratman, 2020-07-29:
Please read the docs.
getMultirotorState and moveBy* apis are what you need. They both take in vehicle_name as an argument.

GeneLYJ, 2020-07-30:
Thank you. Last question, is it possible to implement something like wireless protocol in AirSim? So basically multiple drones communicate through a protocol.

LittleTimeWxk, 2020-08-14:
> Thank you. Last question, is it possible to implement something like wireless protocol in AirSim? So basically multiple drones communicate through a protocol.

Yes，it can realize the function of wireless control. Although I don’t know how to achieve，you can see the video in the docs.You can see it under the title Manuel drive in this link. [https://microsoft.github.io/AirSim/](url)

AIKUUM, 2020-12-24:
> I'm pretty new in AirSim, I managed to create multiple drones in the simulation. I just want to make Drone 1 to say hi to Drone 2. The Drone 2 will reply back to Drone 1 once it received hi from Drone 1.

Hi @GeneLYJ, Were you able to achieve the task? I want to do something similar to what you have done.

GeneLYJ, 2020-12-24:
Hi @AIKUUM, unfortunately, I haven't created nor implemented yet. Since there's no API for wireless communication protocol then I think we will need to write it ourselves. Just only a rough idea of maybe using a formula like Friss Equation for the receiving strength and matrix for the routing table. Eg: A sending signal to the surrounding and any drones receive it with the minimal strength, respond back. And I think you will only able to see the result at the console / Python Shell. This is the simple idea I got. So far I have no better idea or alternate idea.