microsoft/AirSim #2887: car_position = client.getPosition() AttributeError: 'CarClient' object has no attribute 'getPosition'

aldaoctavia, 2020-07-28:
i am trying to run the code testmodel.py program with mountain environment airsim and getting this error 'AttributeError: 'CarClient' object has no attribute 'getPosition',  when i add this code for plotting 

#Additional code for plotting purpose
#    collided = client.simGetCollisionInfo()                                                     
    collided = client.getCollisionInfo()                                                     

    if collided.has_collided == True:
        data = np.column_stack((x,y))
        np.savetxt('d_trial_16.txt', data)
        client.reset()
        break
    
    car_position = client.getPosition()
    #car_position = client.getPosition()
    
    x.append(car_state.kinematics_true.position.x_val)
    y.append(car_state.kinematics_true.position.y_val)
    
    #x.append(client.x_val)
    #y.append(client.y_val)
    print('Sending steering = {0}, throttle = {1}'.format(car_controls.steering, car_controls.throttle))
    client.setCarControls(car_controls)

plt.plot (x, y, color = 'blue', linewidth = 5)
plt.show()

can anyone solve the problem? 

btw, this is my  full testmodel program 

from keras.models import load_model
import sys
import numpy as np
import glob
import os
import AirSim

if ('../../PythonClient/' not in sys.path):
    sys.path.insert(0, '../../PythonClient/')
from AirSimClient import *

# << Set this to the path of the model >>
# If None, then the model with the lowest validation loss from training will be used
MODEL_PATH = None

if (MODEL_PATH == None):
    models = glob.glob('model/models/*.h5') 
    best_model = max(models, key=os.path.getctime)
    MODEL_PATH = best_model
    
print('Using model {0} for testing.'.format(MODEL_PATH))

model = load_model(MODEL_PATH)

# connect to the Airsim simulator
client = CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = CarControls()
print('Connection established!')

car_controls.steering = 0
car_controls.throttle = 0
car_controls.brake = 0

image_buf = np.zeros((1, 59, 255, 3))
state_buf = np.zeros((1,4))

def get_image():
    image_response = client.simGetImages([ImageRequest(0, AirSimImageType.Scene, False, False)])[0]
    image1d = np.fromstring(image_response.image_data_uint8, dtype=np.uint8)
    image_rgba = image1d.reshape(image_response.height, image_response.width, 4)
    
    return image_rgba[76:135,0:255,0:3].astype(float)

x = []
y = []
while (True):
    car_state = client.getCarState()
    
    if (car_state.speed < 5):
        car_controls.throttle = 1.0
    else:
        car_controls.throttle = 0.0
    
    image_buf[0] = get_image()
    state_buf[0] = np.array([car_controls.steering, car_controls.throttle, car_controls.brake, car_state.speed])
    model_output = model.predict([image_buf, state_buf])
    car_controls.steering = round(0.5 * float(model_output[0][0]), 2)

#Additional code for plotting purpose
#    collided = client.simGetCollisionInfo()                                                     
    collided = client.getCollisionInfo()                                                     

    if collided.has_collided == True:
        data = np.column_stack((x,y))
        np.savetxt('d_trial_16.txt', data)
        client.reset()
        break
    
    car_position = client.getPosition()
    #car_position = client.getPosition()
    
    x.append(car_state.kinematics_true.position.x_val)
    y.append(car_state.kinematics_true.position.y_val)
    
    #x.append(client.x_val)
    #y.append(client.y_val)
    print('Sending steering = {0}, throttle = {1}'.format(car_controls.steering, car_controls.throttle))
    client.setCarControls(car_controls)

plt.plot (x, y, color = 'blue', linewidth = 5)
plt.show()

madratman, 2020-07-29:
There is not API called `getPosition`. Please use `getCarState` (https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L1131)

closed 2020-07-29 by madratman

aldaoctavia, 2020-07-30:
i have use getCarState, but then there is another error.
   
File "TestModel.py", line 76, in <module>
    x.append(car_state.x_val)
AttributeError: 'CarState' object has no attribute 'x_val'

[image]

here is my plotting program
[image]
[image]

i want to measure mileage of car when testing in Airsim Simulator, Landscape Mountain Environment.

LittleTimeWxk, 2020-08-06:
I also encountered a similar problem，AttributeError: 'CarState' object has no attribute 'x_val'.
Have you solved it? And what is the reason？

rajat2004, 2020-08-06:
See the `CarState` definition and the fields present in it - https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/types.py#L331-L339