microsoft/AirSim #2888: Blueprint compile error (OptionsMenu)

petergerten, 2020-07-29:
Hi,

it seems this commit https://github.com/microsoft/AirSim/commit/5e873a34ceb751e2c0c5dac135b3f350c03b6244 introduced the following error in my cook process:

```
LogBlueprint: Warning: [Compiler OptionsMenu] Get Assets by Class  was pruned because its Exec pin is not connected, the connected value is not available and will instead be read as default
  LogBlueprint: Error: [Compiler OptionsMenu] The current value of the ' Target Array ' pin is invalid: Array inputs (like 'Target Array') must have an input wired into them (try connecting a MakeArray node). from Source: /AirSim/HUDAssets/OptionsMenu.OptionsMenu
  LogBlueprint: Error: [Compiler OptionsMenu] The current value of the ' Target Array ' pin is invalid: Array inputs (like 'Target Array') must have an input wired into them (try connecting a MakeArray node). from Source: /AirSim/HUDAssets/OptionsMenu.OptionsMenu
```

Using AirSim before this commit works.

rajat2004, 2020-07-29:
There's a PR #2877 which should fix this
Note that I haven't tried it myself

madratman, 2020-07-29:
Yes, this is only happening in 4.25.0. The CI build passed as it's using 4.25.1
PR 2877 does fix it.

saihv, 2020-07-29:
Merged https://github.com/microsoft/AirSim/pull/2877, should fix this.

closed 2020-07-29 by madratman