microsoft/AirSim #2889: libopenvr_api.so issue when building project against UE 4.25

petergerten, 2020-07-29:
When using 4.25 projects that could be build with 4.24 break for me.
@saihv basically the same issue you reported here: https://forums.unrealengine.com/unreal-engine/announcements-and-releases/1755531-unreal-engine-4-25-released/page4

```
[626/627] Link (lld) LandscapeMountains
  ld.lld: error: /UnrealEngine/Engine/Source/ThirdParty/OpenVR/OpenVRv1_5_17/lib/linux64/libopenvr_api.so: undefined reference to std::__throw_out_of_range_fmt(char const*, ...)
  clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

petergerten, 2020-08-13:
I still have this problem with 4.25.
I now verified this with Blocks - and it turns out it only happens in my custom environments for some reason. When I switch back to 4.24 everything works.

@saihv  - as you mentioned a similar issue, do you have any ideas what might cause this?

petergerten, 2020-08-17:
I figured I did not convert the project from 4.24 from 4.25 in the editor yet. I just did that and was hoping to solved the issue - but still the same :-(

petergerten, 2020-08-26:
I am still struggling with this. Basically with latest airsim and UE 4.25.3  I can only get 'Blocks' to work. All other custom environments result in the error above.

@saihv  If you remember what caused the error for you - that could be very helpful :-)

rajat2004, 2020-08-26:
@petergerten I haven't tried packaging with 4.25 so can't say for sure, but if they aren't already, maybe try disabling the VR plugins like being done in [Blocks.uproject](https://github.com/microsoft/AirSim/blob/master/Unreal/Environments/Blocks/Blocks.uproject)

petergerten, 2020-08-27:
@rajat2004 thank you so much - this fixed it. I never though about this as the plugins were not mentioned in my project files. But explicitly deactivating them made the difference :)

closed 2020-08-27 by petergerten