microsoft/AirSim #2893: multithreading buffererror

EdwinMeriaux, 2020-07-29:
Hey!

So I built a ROS publisher to publish the position of all the drones being flown in airsim.
I then built a ROS subscriber with a PID controller to receive this data and then move all of those drones all at the same time.

but now I would like to move the drones in independent threads. but I keep getting the following error:

Exception in thread Thread-508:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "sub1.py", line 67, in run
    client.moveByVelocityAsync(0,0,self.Z_V[self.index],drones.timepause,vehicle_name = self.names[self.index])
  File "/home/edwinm/AirSim/PythonClient/airsim/client.py", line 743, in moveByVelocityAsync
    return self.client.call_async('moveByVelocity', vx, vy, vz, duration, drivetrain, yaw_mode, vehicle_name)
  File "/home/edwinm/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 44, in call_async
    return self.send_request(method, args)
  File "/home/edwinm/.local/lib/python3.6/site-packages/msgpackrpc/session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "/home/edwinm/.local/lib/python3.6/site-packages/msgpackrpc/transport/tcp.py", line 94, in send_message
    sock.send_message(message, callback)
  File "/home/edwinm/.local/lib/python3.6/site-packages/msgpackrpc/transport/tcp.py", line 19, in send_message
    self._stream.write(self._packer.pack(message), callback=callback)
  File "/home/edwinm/.local/lib/python3.6/site-packages/tornado/iostream.py", line 395, in write
    self._write_buffer += data
BufferError: Existing exports of data: object cannot be re-sized

I cannot figure out why this works with 1 drone but not with 2+ drones (why I fly more than 1 I get the error ^^^)

here is my code (the subscriber end):
#!/usr/bin/env python
import sys
sys.path.append('/home/edwinm/AirSim/PythonClient/multirotor') #path to setup_path file
import setup_path 
import airsim
from threading import Thread

import rospy
import numpy as np
import os
import tempfile
import cv2
import math
import time
from std_msgs.msg import Float32MultiArray

#import to read from json file
import json

#imports for graphing
import matplotlib.pyplot as plt

#openning json file
with open("/home/edwinm/Documents/AirSim/settings.json") as f:
    jsonfile = json.load(f)

#opening specific array from json file
#'names' is an array inside of setting.json file for airsim that holds all of the names of the drones
namelist = jsonfile['names']
#'number' is an array insdie of setting.json file for airsim that holds the number of drones
numberofdrones = jsonfile['number']

client = airsim.MultirotorClient()
client.confirmConnection()
for i in range(numberofdrones):
    client.enableApiControl(True,namelist[i])
for i in range(numberofdrones):
    client.armDisarm(True,namelist[i])

bigzerolist = []
bigemptylist = []
for i in range(int(numberofdrones) + 5):
	bigzerolist.append(0)
for i in range(int(numberofdrones) + 5):
	bigemptylist.append([])

class Node(Thread):
	def __init__(self,index, data1, data2):
		self.index = index
		self.Z_V = data1
		self.names = data2
		Thread.__init__(self)
	def run(self):
		print("")
		print("")
		print("")
		print(drones.names)
		print(drones.Z_V)
		print(drones.Z_V[0])
		print(drones.Z_V[0])
		print(drones.Z_V[0])
		print("")
		print("")
		print("")
		client.moveByVelocityAsync(0,0,self.Z_V[self.index],drones.timepause,vehicle_name = self.names[self.index])
		#time.sleep(drones.timepause)

class masterclass:
	#necessary variables
	def __init__(self, numbers, namelist, diffpositionpid,seterror,pause,empty):
		self.number = numbers
		self.names = namelist
		self.Dronedictionary = {}
		self.height = 0
		self.width = 0
		self.diff = diffpositionpid
		self.error = seterror
		self.Ierror = [0,0,0]
		self.Perror = [0,0,0]
		self.Derror = [0,0,0]
		self.Z_diff = []
		self.Z_diff1 = [0,0,0]
		self.tick = 0
		self.timepause = pause
		self.dataplot = empty
		self.classlist = []
		self.Z_V = []
	def PIDcontroller(self):
		print("Ierror",self.Ierror)
		print("Derror",self.Derror)
		print("Perror",self.Perror)
		
		fordataplot = []
		Z_V = [0,0,0]
		print('tick')
		print("newdic",self.Dronedictionary)
		for i in range(self.number):
			if self.Dronedictionary[self.names[i]][1][0] == 0:
				for j in range(1,4):
					print("j",j)
					print(self.Dronedictionary[self.names[i]][1][j])
					self.Dronedictionary[self.names[i]][1][j] = self.Dronedictionary[self.names[i]][0][j-1] - self.diff
				self.Dronedictionary[self.names[i]][1][0] = 1
			self.dataplot[i].append(self.Dronedictionary[self.names[i]][0][2])
		print("newdic",self.Dronedictionary)
		#Z_pos = -1*Z_pos

		#setts up initially Z_diff value(will be change on each loop)
		self.Z_diff = []
		total_Z_diff = 0
		Z_pos = []
		Z_wanted = []
		for i in range(self.number):
			Z_pos.append(self.Dronedictionary[self.names[i]][0][2])
			Z_wanted.append(self.Dronedictionary[self.names[i]][1][2])
			add = self.Dronedictionary[self.names[i]][0][2] + self.Dronedictionary[self.names[i]][1][3]
			self.Z_diff.append(add)
			total_Z_diff = total_Z_diff + add
		print("total_Z_diff)",total_Z_diff)
		#initial print of data
		print("Z_pos: ",Z_pos)
		print("self.Z_diff: ",self.Z_diff)
		if (((total_Z_diff > self.error) or (total_Z_diff < self.error))):
			print("loop entered")
			
			#gains
			Kp = 40 #100
			Ki = 0.01
			Kd = 1

			#BTW Z_diff is error
			for i in range(self.number):
				self.Ierror[i] = self.Ierror[i] + self.Z_diff[i]*self.timepause

			for i in range(self.number):
				self.Derror[i] = (self.Z_diff[i] - self.Z_diff1[i])/self.timepause

			for i in range(self.number):
				Z_V[i] = Kp*self.Z_diff[i] + Ki * self.Ierror[i] + Kd * self.Derror[i]
			self.Z_V = Z_V
			print("Z_V",Z_V)
			
			#velocity movement
			#for i in range(self.number):
			#	self.classlist.append(Node(i))
			#for i in self.classlist:
			#	i.start()
			drone1 = Node(0,self.Z_V,self.names)
			drone2 = Node(1,self.Z_V,self.names)
			#drone3 = Node(2)
			drone1.start()
			drone2.start()
			#drone3.start()
			drone1.join()
			drone2.join()
			#drone3.join()
			time.sleep(0.02)	
			
			#new positioning

			self.Z_diff1 = self.Z_diff
		#adds a list of zeroes for the dictionary positions and the "wanted position"
		#the "wanted position" is the second set of numbers
		#the first number in the "wanted position" list is the set digit
		#if the following values are set then the set digit is 1
	def do_something(self, i):
		return([[0,0,0],[0,0,0,0]])
	#compares relative current and global start positions to make global current position
	def globalposition(self,input):
		listrelative = []
		listglobal = []
		listrelativeglobal = []
		for i in range(self.number):
			listglobal.append([])
			listrelative.append([])
		point = 0
		for i in range(2):
			for j in range(int(self.width/2)):
				for k in range(self.height):
					if i == 0:
						listrelative[k].append(input[point])
					elif i == 1:
						listglobal[k].append(input[point])
					point = point + 1
		for i in range(self.number):
			local = []
			for j in range(len(listrelative[i])):
				local.append(listglobal[i][j] - listrelative[i][j])
			listrelativeglobal.append(local)
		for i in range(self.height):
			self.Dronedictionary[self.names[i]][0] = listrelativeglobal[i]

		#self.GRP = listrelativeglobal

	#standard callback for the published data
	def callback(self, data):
		print("")
		print("")
		rospy.loginfo(rospy.get_caller_id() + "I heard %s", data.data)
		r = rospy.Rate(5)
		val1 = data.layout.dim[0].stride
		val2 = data.layout.dim[1].stride
		self.height = data.layout.dim[1].size
		self.width = data.layout.dim[0].size
		print("val1",val1)
		print("val2",val2)
		print("h",self.height)
		print("w",self.width)
		self.globalposition(data.data)
		print("dict142",self.Dronedictionary)
		if self.tick <= 250:
			self.PIDcontroller()
		else:

			for i in range(self.number):
				plt.plot(self.dataplot[i],label = "drone" + str(i))
			 
			plt.xlabel('increment number') 
			 
			plt.ylabel('drone position (height)') 
			plt.legend() 
			plt.show()
		
		self.tick += 1
		print("TICK",self.tick)
		#r.sleep()
	#main fuction just calls callback()
	def main(self):
		Dronedict = {"Drone" + str(i + 1) : self.do_something(i) for i in range(numberofdrones)}
		self.Dronedictionary = Dronedict
		rospy.init_node('listener', anonymous = True)
		rospy.Subscriber("sent_matrix",Float32MultiArray,self.callback)
		rospy.spin()

if __name__ == "__main__":
	drones = masterclass(numberofdrones,namelist,10,0.1,0.01,bigemptylist)
	drones.main()

would anyone know why this error occures?

closed 2020-08-02 by EdwinMeriaux

AlibekJ, 2022-07-30:
Hey @EdwinMeriaux, have you managed to solve this? How? Please do tell us.

ME-Msc, 2024-05-08:
I have tried a lot of method to solve the problem of mulit-threading to control several vehicles in Airsim. The best way is to create different client for each vehicle.
```
uavs_name_list = ["uav_0", "uav_1", "uav_2"]
clients = {}
for uav_name in uavs_name_list:
	client = airsim.MultirotorClient()
	clients[uav_name] = client
	clients[uav_name].confirmConnection()
	clients[uav_name].enableApiControl(True)
	clients[uav_name].takeoffAsync().join()
```