microsoft/AirSim #2907: Lidar in computer vision mode

seshanrv, 2020-08-03:
Is it possible to add a lidar sensor in computer vision mode? From the documentation and after a few trials, I see that the getLidarData() and the simGetLidarSegmentation() (and probably many other VehicleClient python APIs) are not available in computer vision mode. Has anyone tried doing this? 

My objective is to capture camera images and lidar point clouds for a vehicle with parts that have a working function (for e.g. a boom). I want to define the physics externally and only update the position of the vehicle and the boom using simSetVehiclePose(). The idea is to define the vehicle and the boom as seperate computer vision "vehicles". I am able to attach a static mesh of the vehicle body to the computer vision pawn blueprint in UE4Editor, that makes it look like the camera is attached to the vehicle. But haven't succeeded in inserting multiple such CV pawns (for the boom) apart from being unable to use lidars. 

Does anyone know if there is an easy way to get this done? Thanks a lot in advance. 

PS: I already tried URDFSim. It is great work by @mitchellspryn , but since my project is going to run for a while I really want to stay updated with the main branch of Airsim.

ale99viv, 2020-11-20:
Did you found a solution for using lidar sensor in computer vision mode? I have the same problem

QiaoRuihuan, 2023-08-29:
I also encountered the same problem, is there a solution?