microsoft/AirSim #2910: Confusion on general workflow

Batman-wayne, 2020-08-04:
Hello, 

I am very new to using AirSim. I am using the recommended setup, using VS 2019 on Windows 10 and Unreal Engine 4.24. I was able to make my own custom Unreal environment and run the car just fine, but I am confused on how to proceed from here. I am using C++ and I am trying to use the APIs to control the car, but I have no idea how to proceed. I thought I would add a .cpp and .h file and then use the APIs, but it seems I have to do this from Visual Studio. Is there a way to add it in such a way that in generates the .vxproj files? Also, I followed the general workflow and attempted to add a file, then use the bat files as recommended [here](https://microsoft.github.io/AirSim/dev_workflow/), but now I cannot run the original project. What am I misunderstanding? Any help would be appreciated!

rajat2004, 2020-08-04:
I haven't worked with C++ APIs much even less on Win, so not sure as to how to generate the project files
You could have a look at [`HelloCar`](https://github.com/microsoft/AirSim/tree/master/HelloCar) or [`HelloDrone`](https://github.com/microsoft/AirSim/tree/master/HelloDrone)

jonyMarino, 2020-10-09:
Hi @harris-mohamed. The simplest way to do it is just to copy paste the HelloCar project, change its name and add it to the AirSim solution.
Tell me how it went!

closed 2020-11-23 by jonyMarino