microsoft/AirSim #2911: communication with custom unreal engine

lucafei, 2020-08-04:
My configuration:  window 10 , unreal engine 4.24, airsim latest, 

hello, guys, i have created a own unreal environment according to the tutorial in https://microsoft.github.io/AirSim/unreal_custenv/.   After i finished the unreal environment, i runed the environment und tried to use hello_drone.py in https://github.com/microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py, but nothing happened. I don't know where i should make some configuration? Does anyone have some ides for that? Thank you in advance!

xxEoD2242, 2020-08-04:
Do you have any type of anti-virus software actively installed?

rajat2004, 2020-08-04:
Is the drone appearing as normal when you press play? Or able to drive the car in the env.
A missing step could be setting `GameMode Override` to `AirSimGameMode`

lucafei, 2020-08-04:
> Is the drone appearing as normal when you press play? Or able to drive the car in the env.
> A missing step could be setting `GameMode Override` to `AirSimGameMode`

yes, the drone has already appeared and i have set the mode in AirSimGameMode

lucafei, 2020-08-04:
> Do you have any type of anti-virus software actively installed?

i think it is only windows defender

rajat2004, 2020-08-04:
And `airsim.confirmConnection()` is working?

lucafei, 2020-08-04:
> And `client.confirmConnection()` is working?

yes, it's working, just can not control

lucafei, 2020-08-04:
do i need to make some codes in myclass of unreal project?

xxEoD2242, 2020-08-04:
> > Do you have any type of anti-virus software actively installed?
> 
> i think it is only windows defender

I had similar issues before and it was due to the fact that the anit-virus sofware was preventing the msgpack server from receiving messages from my shell. Try turning off Windows Defender and see what happens.

lucafei, 2020-08-07:
> I had similar issues before and it was due to the fact that the anit-virus sofware was preventing the msgpack server from receiving messages from my shell. Try turning off Windows Defender and see [...]

Hi, @xxEoD2242 it is now working, and i have a question for the drone control, can the drone be controled in airsim directly with keyboard?  because i tried to control drone with keyboard but it doesn't work. Moreover did you try to control drone with remote controller from DJI, which i want to use in airsim?

xxEoD2242, 2020-08-07:
> Hi, @xxEoD2242 it is now working, and i have a question for the drone control, can the drone be controled in airsim directly with keyboard? because i tried to control drone with keyboard but it [...]

To my knowledge, currently, the drone can only be controlled by using a controller. I believe the AirSim documentation details how to connect this device. I haven't personally worked with controllers and have focused more on Python APIs.

jonyMarino, 2020-09-02:
This became a dup of #2666

closed 2020-09-02 by jonyMarino