microsoft/AirSim #2913: Is it possible to implement Wireless Protocol in AirSim?

GeneLYJ, 2020-08-05:
Hi, I've tried looking for a way to implement wireless communication among multiple drones but have no result. I'm wondering if there's custom API or any ways to implement wireless communication in AirSim? Many thanks.

rajat2004, 2020-08-07:
What do you exactly mean by Wireless communication?
As you probably have already seen, APIs allow usage for multiple vehicles, and ROS is a very popular way of working with multiple robots

GeneLYJ, 2020-08-07:
Wireless Communication as in something similar to Bluetooth. Is ROS widely used in AirSim?

xxEoD2242, 2020-08-07:
@GeneLYJ I have implemented a custom messaging protocol in AirSim that utilizes the standard mspack system that AirSim is built upon.

If you wanted to do something like Bluetooth, you would need to write the messaging protocols yourself. I got around this during academic research by emulating the drone communications utilizing a matrix and some basic distance metrics. If you're interested, I can send you the fork of AirSim where I did some of this work.

GeneLYJ, 2020-08-08:
Yes that would be great. Thank you so much!