microsoft/AirSim #2916: Collisions with animals in NH environment

HemaZ, 2020-08-05:
Hello, 
I've noticed a strange behavior in the neighborhood environment. sometimes I find that the simulated car is hit aggressively and flying in the sky !!. 
 I've noticed that when a deer hit the car while driving it changed the direction. so I'm suspecting that the animals are hitting the car and causing this behavior. 

please check these 2 videos.

https://youtu.be/g_r3RE1LzBk

https://youtu.be/oi2grTE_G4E

rajat2004, 2020-08-07:
Yes, the animals seem to be a new feature in the Neighbourhood environment. If have access to the source code of the env, could remove the actors. There are also new APIs in the master branch such as `simDestroyObject` which can remove the actors specified, this will be present in the next release

GimpelZhang, 2020-08-25:
My car got hit by the animals as well when I used the NH environment. 

Since I didn't have the source code of the env, I decided to make one by myself. I put it in the release of my repo: 
[GimpelZhang/airsim_ros](https://github.com/GimpelZhang/airsim_ros). Hope that will help if you are still working on it.

HemaZ, 2020-08-25:
> My car got hit by the animals as well when I used the NH environment.
>
> Since I didn't have the source code of the env, I decided to make one by myself. I put it in the release of my repo: [...]

Thank You. im defiintely giving it a try

closed 2021-08-19 by jonyMarino