microsoft/AirSim #2921: How to obtain the data information of all vehicles in the multi-vehicle state？

LittleTimeWxk, 2020-08-10:
Dear friends，
    Now I create many cars in simulator and I want to put the datas of them into a list, then I can get the information of the car one by one using python API.
    
     **My Question：How to add all the vehicles created to a data list？**

     I don't understand the code structure of vehicle creation in codes, so I won't use it.Could you provide some help？Tutorials, suggestions and ideas are all available！

rajat2004, 2020-08-10:
Most of the APIs take in a `vehicle_name` parameter, which should be the name specified on the settings
Not passing the parameter uses the first vehicle

LittleTimeWxk, 2020-08-10:
> Most of the APIs take in a `vehicle_name` parameter, which should be the name specified on the settings
> Not passing the parameter uses the first vehicle

Thanks your reply. But I think, may be you can't understand my question.

rajat2004, 2020-08-10:
Well, then maybe you should have explained it better
See the documentation - https://microsoft.github.io/AirSim/multi_vehicle/

LittleTimeWxk, 2020-08-10:
> Well, then maybe you should have explained it better
> See the documentation - https://microsoft.github.io/AirSim/multi_vehicle/

OK! Thank you for your reminder.

LittleTimeWxk, 2020-08-12:
> Well, then maybe you should have explained it better
> See the documentation - https://microsoft.github.io/AirSim/multi_vehicle/

Hi! I have read the documentation, but that's not what i want. I want get the status of all the car one-time. 
Now my plan is:
step1: I can get the name of all control car, and put it to a list;
step2: I can get the status of the corresponding vehicle by name list, and put status to a new list;
step3: I will post the state list to a ros point.

My problem is I don‘t know how to get all the names of the car in C++ backend in the first step. Because There are many things I don’t understand in the code project.
Could you give me some advice and other ideas?

rajat2004, 2020-08-14:
Well, you're writing the settings file, so you would already know what are the names of the vehicles. An advanced step could be to use a JSON parser and parse the settings yourself. 
This does bring up a possibility of adding a new API to give a list of the vehicles, which could be useful especially if and when https://github.com/microsoft/AirSim/pull/2390 gets in (needs update and lots of testing though)
If you're using ROS, have a look at https://microsoft.github.io/AirSim/airsim_ros_pkgs/ also, it might contan what you need

LittleTimeWxk, 2020-08-20:
> Well, you're writing the settings file, so you would already know what are the names of the vehicles. An advanced step could be to use a JSON parser and parse the settings yourself.
> This does bring [...]

Thank you very much! I have solved it just now.

closed 2020-08-20 by LittleTimeWxk