microsoft/AirSim #2922: Allocate the number of drones after pressing "Play" instead of the setting in advance?

saeed97, 2020-08-10:
1-	We are working on a research project, and we use more than 1000 drones in the Unreal Engine. Using more 1000 drones takes memory, and we have to identify the number of drones in the JSON setting file before pressing “Play” button even if we did not use all of them. Is there a way to automate or allocate the number of drones after starting the simulation/ after pressing play? 
2-	Passing a specific command to a drone such as “moveonpath” is being canceled by passing another command before finishing executing the first command. I need to control the velocity by changing it while it is still moving on the specified path from the previous command. Is there a way to do a combination of APIs. For example, it moves on the path and passes another command to increase its speed without canceling the previous command?  
3-	Is there a way to use different drone vehicle bodies such as fixed-wing drones?

LittleTimeWxk, 2020-08-14:
For the first question， You can set the number of vehicle in the JSON and modify the loading method of the vehicle in the C++ backend. But the most important is that you must understand the code for parsing json and loading vehicles. You have to read the source code carefully.
I hope my suggestion can help you!

rajat2004, 2020-08-14:
1. Currently, there's no way of creating vehicles as runtime, there's a PR https://github.com/microsoft/AirSim/pull/2390, but needs rebase and lots of testing. I guess currently you're maybe changing the number of vehicles in a script, then creating the settings file from there.
2. From a very brief look at the code, cancelling the previous command shouldn't happen, but needs looking into
3. Not as of now, there are several related issues which discuss about this possibility and how it could be added

closed 2021-03-09 by zimmy87