microsoft/AirSim #2923: Using Lidar in Training

burak617, 2020-08-11:
I want to use lidar to train RL model. However lidar data's size is different depending on the situation. I need to fixed-sized lidar vector to feed my network. How do I overcome this situation?

EivindYN, 2020-08-11:
I found a fix that makes the lidar send the full scan everytime, making it fixed-sized (https://github.com/microsoft/AirSim/issues/1997)

ivannson, 2020-09-18:
You can work around the issue by figuring out which points are missing and adding them into the scan. Each point has a unique vertical angle - horizontal angle combination, sometimes points are missing so no value is return. You can figure out at which angle values the points are missing and reinsert them with the maximum range value (i.e. as if the point was detected far away). This is similar to applying a mask when converting a lidar scan into a depth image, which is also something you may want to look into, as each image has a set size regardless of the number of points.