microsoft/AirSim #2924: Airsim crashes if you run px4 and close airsim before 2 seconds.

EivindYN, 2020-08-11:
Reproducing steps:
* Run px4
* Run and exit AirSim before 2 seconds passes

Errors (and fixes):
[image]
Then after fixing this error with the commented out line you get this:
[image]
This naturally happens as mav_vehicle = nullptr after disconnecting, so using the commented out lines fixes this too.
* Run and exit AirSim before 2 seconds passes* Run and exit AirSim before 2 seconds passes

lovettchris, 2021-06-04:
I may have fixed this bug in this pull request:
https://github.com/microsoft/AirSim/pull/3757

zimmy87, 2021-06-16:
This issue doesn't repro for me w/ AirSim master connecting with PX4 master running over WSL2 and PX4 v1.11.3 running over cygwin. This is before I applied the fix in #3757.

PR #3757 (2021-06-18): Fix bug in UDP socket failing with WSAECONNRESET in the read thread

Fixes: #3698    
Fixes: #2924

## About
Fix bug in UDP socket failing with WSAECONNRESET in the read thread if it tries to read to soon.  The fix is to put a retry count in the UdpClientPort where it automatically reconnects up to 10 times.  Also fixed a null ref crash on mav_vehicle_ that happens in some cases like if you kill px4 process while drone is flying.

## How Has This Been Tested?
Windows with PX4 in WSL2 and in WSL1 with latest PX4 master bits as of today (1.12.0)

## Screenshots (if appropriate):
closed 2021-06-18 by zimmy87