microsoft/AirSim #2925: AirSim crashes when requested an image

marco-stephan, 2020-08-11:
When requesting an image from AirSim using `simGetImage(...)` or `simGetImages(...)`, AirSim occasionally crashes. This only happens on one of my computers.

## simGetImage(...)
### Reproduction steps
1. Connect to AirSim
2. Request image(s)

Code:
```C++
MultirotorRpcLibClient* airSim = new MultirotorRpcLibClient();
while (airSim->getConnectionState() != RpcLibClientBase::ConnectionState::Connected);

while (1)
{
	const std::vector<uint8_t> image = airSim->simGetImage("0", ImageCaptureBase::ImageType::Scene);
}
```

### Error
[image]
DelegateSignatureImpl.inl at line 1045

The data of the `DelegateInstanceInterface` variable is unaccessable ("Unable to read memory")

## simGetImages(...)
### Reproduction steps
1. Connect to AirSim
2. Request image(s)

Code:
```C++
MultirotorRpcLibClient* airSim = new MultirotorRpcLibClient();
while (airSim->getConnectionState() != RpcLibClientBase::ConnectionState::Connected);

while (1)
{
	const std::vector<ImageCaptureBase::ImageRequest> imageRequests = {
		ImageCaptureBase::ImageRequest("0", ImageCaptureBase::ImageType::Scene, false, false)
	};
	const std::vector<ImageCaptureBase::ImageResponse>& imageResponses = airSim->simGetImages(imageRequests);
}
```

### Error
[image]
RenderRequest.cpp at line 84

The `game_viewport` variable is of value `0x0000000000000001` and its data is unaccessable ("Unable to read memory")

## Software Information
AirSim commit: bd8ee6b (latest master)

## Hardware Information
CPU: Intel i5-7300HQ
GPU 1: Intel HD Graphics 630
GPU 2: NVIDIA GTX 1050Ti
RAM: 8GB

rajat2004, 2020-08-11:
Which version of UE are you using? And this is happening only on Windows? (I'm assuming here that multiple systems includes different OS also)
The first one is happening from inside UE code, so can't really do much about that, maybe a later version can help
As for the second one, that doesn't make much sense, `game_viewport_` is being used multiple times before this as well in the same function, so if it's null, should be exited in the beginning itself

marco-stephan, 2020-08-11:
I am using UE 4.24.3 on both computes and both machines are running on Windows 10 (10.0.19041.388).

I was able to fix both crashes in https://github.com/marco-stephan/AirSim/tree/fix-get-images. Note that I have very little experience with AirSim's source code and the Unreal Engine itself so the fix committed is most likely not optimal.

rajat2004, 2020-08-11:
The crash was somewhat reproducible and this fixed the problem? There's an already existing `wait_signal_` which does the same thing, so instead of adding a new one, it might be better to move the call of `wait_signal_->signal()` from the end of `Execute` to the end of the lambda function where you right now added `wait_signal2_->signal()`
Anyways, a PR would be great

marco-stephan, 2020-08-11:
The crashes were pretty reproducible meaning they appeared within 2 to 3 seconds when executing the code above. I wasn't sure if the proposed fix is the right way to go so I did not open a PR for this. That being said, I will open a PR with only one signal.