microsoft/AirSim #2931: Pip Airsim package outdated

agentfeux, 2020-08-13:
Hi,
I was working with the Airsim pip package until recently when i did the recent update. Some function name have changed so it's doing some conflict with older code. I was getting some wierd error and fortunately i found quickly why.

Would be nice if someone update it :P

Thanks you !

rajat2004, 2020-08-13:
A new package will be released only with a new release I think. I'm using an editable pip install, `pip install -e .` from PythonClient folder
As for changed API, are you referring to `simSetCameraOrientation`? That was recently upgraded to `simSetCameraPose` but a backwards-compatible layer wasn't added, which now that I think about it should be done, with some warning messages

agentfeux, 2020-08-19:
Yes, I was referring to this function. Thank you I didn't know I was able to do that with 
pip !

closed 2020-08-19 by agentfeux