microsoft/AirSim #2940: How to map AirSim actions to steering wheel buttons?

ahmedkrmn, 2020-08-16:
I'm using the [RawInput Plugin](https://docs.unrealengine.com/en-US/Gameplay/Input/RawInput/index.html) to add Unreal action mappings for a generic steering wheel. I'm using the Engine-Input UI settings menu for doing so.

I want to also map AirSim specific actions to the steering wheel as well. 

**For example, "Recording" into the log file would be mapped to "GenericUSBController Button 7" as shown below:**

[image]

**So does AirSim provide a list of unique mapping names that can be used for achieving this?**

_Thanks in advance._

jonyMarino, 2022-03-31:
https://github.com/saihv/AirSim/blob/master/Unreal/Plugins/AirSim/Source/ManualPoseController.cpp
Closed due to inactivity. Please, feel free to ask for it to be reopened if you wish to continue posting.

niwc-jamelc, 2022-04-18:
@jonyMarino I'd like to reopen this issue as a feature request to add the UE4 editor Input Mapping system in UE4's Project Settings and Config ini file.

Making changes to any of the inputs required digging through the plugin code and recompiling it anytime you wanted to make a keybinding change. Also, any Keybinding changes would not take effect on the HUD.

I've been working on a branch that allows users to remap the default controls to any keyboard and gamepad key, fixed movement issues with the Manual Camera controls, fixed the mislabeling of the car pawn's forward and backward cameras, and to automatically update the HelpText with the new keybindings in the HelpHUD and Weather Options Menu.

While I don't have a steer wheel controller to test the raw input plugin, the change to using the Input Mapping system should be compatible with usage of the Raw Inputs plugin.

I'd like to submit a Pull Request with these changes and bug fixes.