microsoft/AirSim #2945: How to edit CityEnviron map and AirSim plugin configurations?

GilmarCorreia, 2020-08-18:
I and a group of friends are trying to create an Autonomous Racing Car Challenge using AirSim and CityEnviron map. We've made some progress trying to automize some part of the code, but we want to add a StopWatch in the HUD, with start and finish line. We've made something like the picture below: 

[image]

We are facing some trouble putting this structure in the CityEnviron map. How can I open the CityEnviron in Unreal Engine using the .pak files? Or how could I edit the AirSim plugins? If you have some idea, please share with us :) 

OS: Windows
AirSim version: 1.3.1
Unreal: 4.24

rajat2004, 2020-08-18:
Looks pretty good!
The binaries can't be edited AFAIK, you'll have to use the source code of the env, can be bought from the Unreal Marketplace

denmonz, 2022-08-26:
@rajat2004 do you happen to know the name of the environment in the Unreal Marketplace?