microsoft/AirSim #2951: Orientation and position of Lidar/Camera attached to the Car

thejeshk, 2020-08-19:
Hello AirSim Community, 

Can we change the orientation and position of the Lidar sensor or Camera attached to the Vehicle (Car)

Thank you 
TK

rajat2004, 2020-08-19:
See the documentation - https://microsoft.github.io/AirSim/lidar/, https://microsoft.github.io/AirSim/settings/
They both take X,Y,Z & Roll,Pitch,Yaw as parameters in settings

closed 2020-08-26 by jonyMarino

riyadfebrian, 2021-06-18:
is there any explanation to set X, Y, Z for dummies ?

jonyMarino, 2021-06-18:
Hi Riyad! read the docs please: https://microsoft.github.io/AirSim/lidar.html#lidar-configuration

riyadfebrian, 2021-06-19:
Thanks, Mr. JonyMarino. but what I mean is basic knowledge about X Y Z in 3d Environment AirSim. I know its out of coverage documentation, but if the documentation has some reference about it, I think it'll very useful especially for beginners.

or is there any example setting up position of sensor ? 

i.e distance sensor in right side of vehicle , set X=0, Y=1, Z=-1

jonyMarino, 2021-06-21:
We use NED always:
https://microsoft.github.io/AirSim/apis.html#coordinate-system