microsoft/AirSim #2954: macos AirSim + PX4 + ROS

KristofferFogh04, 2020-08-20:
Hello!

I am not sure where to post this question, so please remove if not appropriate.
I am a beginner using AirSim, and intend on using it for a research projet in drone racing at my university.

I have AirSim running together with PX4 and it works great. I can control the drone from PX4 and define missions in QGroundControl and the drone will fly in AirSim UnrealEngine with a decent fps. The next step is then to use it with the ROS wrapper. The issue I have is that i'm running AirSim and PX4 in macos (on a Macbook Pro 15' 2018), and as far as I can see the ROS adapters (mavros etc.) is not supported on macos. I have tried running ROS on a ubuntu VM but I cant seem to connect with AirSim and PX4 running in macos.

I have looked at this similar issue, and tried to edit the IP-address to the host machine, but I cant seem to get it working https://github.com/microsoft/AirSim/issues/2850

So my question is, have anyone gotten this kind of setup working, and is it even possible? If yes, can you supply some kind of guide as to how to setup these connections?
The alternative would be to run everything in the ubuntu VM, but I assume i'll run into some performance issues.

I hope this all makes sense, and thank you in advance. Best regards

rajat2004, 2020-08-20:
At first glance, I don't see any reason why it shouldn't work, other than the settings parser not working due to file being on a different system. Not exactly sure how it would affect the ROS wrapper though.
Any kind of errors which you're getting? Are the machines reachable by each other on the IP addresses being used?

KristofferFogh04, 2020-08-20:
Thank you for the reply. Not getting any errors. Regarding if they are reachable, I have only tried pinging my host IP to the VM IP and the other way around and they seem to be reachable that way. I should probably look closer into it with wireshark or something, so that's the next order of business.
When I think about it, it might also just be because I simply don't fully understand the ROS architecture and how I should apply it. Should I have both the airsim_ros_pkgs node and mavros node running on the ubuntu VM? I have tried this where I set up mavros with the UDP port and address that PX4 accept with no luck. 

Anyways, all general guidance in what ROS nodes should be running and how they should be set up is very much appreciated. Thank you again.

rajat2004, 2020-08-21:
I think everything ROS related should be the Ubuntu VM. How did you launch Mavros with the IP & Port of PX4, using something like `roslaunch mavros px4.launch fcu_url:="udp://:14540@192.168.1.36:14557"` ? Try to first make sure that Mavros connects with PX4, then bring Airsim ROS nodes into the picture
I would also suggest trying running just AirSim in MacOS, & PX4 + ROS in Ubuntu VM, however, there have been reports that running PX4+AirSim on different systems doesn't work, so not sure.

Yueting-Li, 2020-10-06:
Hey, i am currently using Airsim on the Windows10, and i want to run px4 + ros + mavros on another machine(ubuntu). So I use a cable to connect the two computers. I set Windows ip address as 192.168.1.1, and ubuntu ip as 192.168.1.2. 

Then I successfully run PX4 on ubuntu machine and connected to Airsim. But when I tried to bring Airsim ROS nodes into the picture, it doesn't work. I just changed "host_ip"(in airsim_node.launch) to windows IP address 192.168.1.1.
Firstly I runned Airsim on windows, then I runned airsim_node.launch on ubuntu. The terminal showed "Waiting for connection - XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"
Can anybody give me some suggestions?
Maybe I did something wrong. Thanks a million!