microsoft/AirSim #2955: Fixed Wing Aircraft

AOS55, 2020-08-20:
# Fixed Wing Vehicle Integration

I am working on adding a FixedWing aircraft class into Airsim, so far I have been documenting my work under issue #2508 but as suggested by @rajat2004 it would be usefult to move this work to a new issue so as to improve visibility for others to work on. 

### Work so far
Please see this [post](https://github.com/microsoft/AirSim/issues/2508#issuecomment-674392713) for a detailed breakdown of what has been added in my [forked Airsim repository under the fixedwing branch](https://github.com/AOS55/AirSim-1/tree/FixedWing). I will aim to keep updates on this issue of new files, classes, methods & functions with a view to write comprehensive documentation as the code begins to get to a state where it can be merged back into the master branch on the main repo. 

Since the [last update post](https://github.com/microsoft/AirSim/issues/2508#issuecomment-674392713) on #2508 I have been updating the FixedWingLibClient files to enable communication with Ardupilot. A linked issue has been opened within the [ArduPilot Repo](https://github.com/ArduPilot/ardupilot/issues/15095). 

### FixedWing Branch State
- Compilable from Visual Studio: 🔴 
- Runs alone in UE4: 🔴 
- Runs with ArduPilot and Mavlink in UE4 over network: 🔴 
- Model validated with real comparison: 🔴 

### Known issues
- I can't compile the code from source in Visual Studio please see the [chain](https://github.com/microsoft/AirSim/issues/2508#issuecomment-675355753) back in #2508 for what we have tried so far. I think it is most likely a mistake I have made in the source code. <-- This is currently the most pressing issue.
- I naievly initially implemented the control surfaces at seperate functions `getElevatorSignal()` etc. I am beginning to realise as I work throught the control implementation this was not a good way to go about this and the program is already well setup to accept a more conventional control vector, i.e. the **u** vector in state-space form. I am planning to go back and correct this to a 4 row **u** vector.
- The Core APIs I effectively copied over when setting up the controller files include several aspects that I don't think make sense for a Fixed Wing aircraft, for example [commandVelocityHold](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/firmwares/mavlink/MavLinkFixedWingApi.hpp#L551) uses the function `moveByLocalVelocity` that takes a boolean `yaw mode` switch and `rate` as a command these are set as `false` and `0` in the current implementation. But, `moveByLocalVelocity` may be better of if rewritten for fixedwing. 

### Planned Future Work
Initially I am aiming to work to resolve all the known issues and turn all the 🔴 in the state to 🟢. Following from this I will aim to add the following functionality:
[ ] Construct a similar Python API as in multirotor <-- hopefully a short term objective
[ ] Implement Undercarriage for fixedWingLandings
[ ] Deep Reinforcement Learning (DRL) based FW landings <-- main PhD objective, expect to write & publish a paper based on this
[ ] Validate model and control on real UAV and FW aircraft, ties in with above. 

### Motivation
Principally this is for my PhD at the [University of Bristol](https://www.bristol.ac.uk/) where I am aiming to use a combination of computer vision and Deep Reinforcement Learning to train an autonomous fixedwing agent/aircraft to navigate to and safely land in an unknown location. This is both with and without power and on a range of fixedwing aircraft including fixedWing UAVs, gliders & multiengine light piston aircraft (<5700Kg). Not only this but when discussing AirSim with my colleagues we believe that having a fixedWing simulator capable of highfidelity graphics and SITL that is efficient will be of great benefit. 

### Ongoing Development
Several users have already been realy helpful (@rajat2004 😀), if anyone has any ideas about how to solve some of the issues I have described above all help is greatly appreciated. In particular if anyone is able to have any success compiling the [fixedwing source](https://github.com/AOS55/AirSim-1/tree/FixedWing) please let me know as I'm a little stumped 😕 as to why it won't compile.

rajat2004, 2020-08-21:
I had a try at compiling your branch, the error message was really quite useless.
Opened https://github.com/AOS55/AirSim-1/pull/1 which fixes that and brings us back to normal compilation errors with understandable messages (for now)

AOS55, 2020-08-24:
### Progress Update

#### Aircraft Physics Control Construction
I have been continuing to work on the FixedWingPhysics simulation aiming to resolve compilation errors. I have vectorised the control surfaces so ideally they behave as a **u** tyoe vector. With this I introduced a new file [Aeroplane.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/Airplane.hpp) that instants each control from the [ControlSurface.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/ControlSurface.hpp) file as a PhysicsBodyVertex class. I have not introduced the thrust vector yet but this shouldn't be too much of a challenge. 

Aircraft links to the structs in [AircraftParams.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/AircraftParams.hpp) and applies the aero-forces via `setWrench`. Most fixedwing dynamic models/databases, define the forces and moments around the CG, as `dynamicDerivatives`. setWrench applies these and then is updated in the loop by [FixedWingPhysicsBody.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/FixedWingPhysicsBody.hpp), I'm a little lost as to how the PhysicsBody class updates the aircrafts pose via `fastPhysicsEngine`. The way I've set up the classes may also be an antipattern and I might need to refactor some of the structs, please let me know your thoughts on how you might setup the structures. 

#### Compilation error
When I run the repo from visual studio I get the following error :

`Error C4717 'msr::airlib::ArduPlaneApi::resetImplementation': recursive on all control paths, function will cause runtime stack overflow Blocks C:\Users\quessy\Documents\Unreal Projects\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\vehicles\fixedwing\firmwares\ardupilot\ArduPlaneApi.hpp	57`

Im not sure what this error is being caused by, I'm guessing there is no termination condition in the recursive call. If anyone has experienced this before I'd love to get your thoughts.

rajat2004, 2020-08-24:
That's happening cause the function is calling itself here - https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/vehicles/fixedwing/firmwares/ardupilot/ArduPlaneApi.hpp#L54
It should instead be calling `FixedWingApiBase::resetImplementation()`, from it's parent class

AOS55, 2020-08-24:
Thanks yep that was it was quite obvious now I think about it in more detail. It several unresolved external symbol errors now but I think there a case of sorting out further problems with the API.

AOS55, 2020-08-24:
Does anyone have any thoughts on this error coming from `FixedWingApiBase.hpp`, I thought it was to do with the link between the base class and Mavlink/Px4 implementation override. I can't see anything untawards though:
LNK2001	unresolved external symbol "protected: virtual float __cdecl msr::airlib::FixedWingApiBase::getAutoLookahead(float,float,float,float)const " (?getAutoLookahead@FixedWingApiBase@airlib@msr@@MEBAMMMMM@Z) Blocks C:\Users\quessy\Documents\Unreal Projects\AirSim\Unreal\Environments\Blocks\Intermediate\ProjectFiles\Module.AirSim.cpp.obj	1	

[image]

rajat2004, 2020-08-24:
Functions seem to be missing, are all the files uncommented in the [AirLib.vcxproj](https://github.com/microsoft/AirSim/compare/master...AOS55:FixedWing#diff-0e5e296b5734ca4828f774cf0a9bc67e)?

AOS55, 2020-08-24:
Thats a good point I've uncommented them now, from [AirLib.vcxproj](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/AirLib.vcxproj), but Im still getting the same link errors from the compiler. When running build.cmd I also get some errors in the command prompt that are slightly different, perhaps these are what needs to be looked at. 
[image]

AOS55, 2020-08-24:
Resolved all build errors now except for one rather unusual one:
`"C:\Users\quessy\Documents\Unreal Projects\AirSim\AirSim.sln" (default target) (1) ->
"C:\Users\quessy\Documents\Unreal Projects\AirSim\AirLib\AirLib.vcxproj" (default target) (12) ->
 (ClCompile target) ->
C:\Users\quessy\Documents\Unreal Projects\AirSim\AirLib\deps\rpclib\include\rpc\msgpack\v1\object.hpp(202,11): error C2039: 'msgpack_pack': is not a member of 'Eigen::Matrix<float,3,1,0,3,1>' [C:\Users\quessy\Documents\Unreal Projects\AirSim\AirLib\AirLib.vcxproj]`

rajat2004, 2020-08-25:
The RPC error was due to the fact that it doesn't know how to pack the Vector3r object, need to convert it into the struct defined in RpcLibAdaptorsBase and then pass it. Opened https://github.com/AOS55/AirSim-1/pull/4 which fixes that, and adds the files to CMake.
After this AirLib compiles, but there are still lots of errors when building the entire Blocks env, undefined references in FixedWingApiBase.cpp, due to function declarations in .hpp, and usage of commented out methods

AOS55, 2020-08-25:
Great thanks very much Ill take a look at those errors today, I merged the issue back into fixedwing and have no build.cmd errors on my end too.

AOS55, 2020-08-25:
Any thoughts on why I'm getting the following include/visibility related build error from in visual studio? I can see the source file [UdpSocket.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/MavLinkCom/include/UdpSocket.hpp) is still on the repo, could it be related to cmake changes?

- From Visual Studio: `C1083 Cannot open include file: 'UdpSocket.hpp': No such file or directory Blocks C:\Users\quessy\Documents\Unreal Projects\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\vehicles\car\firmwares\ardurover\ArduRoverApi.hpp`

- From Output `C:\Users\quessy\Documents\Unreal Projects\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\vehicles/car/firmwares/ardurover/ArduRoverApi.hpp(20): fatal error C1083: Cannot open include file: 'UdpSocket.hpp': No such file or directory`

rajat2004, 2020-08-25:
That's very strange, Windows shouldn't be affected AFAIK since it uses the .sln files. And if this were a problem, then it would have appeared way earlier since support for Rover has been present for many months
Does this appear when running `build.cmd`?

AOS55, 2020-08-25:
No thats the unusual thing build.cmd doesnt present any errors when run

rajat2004, 2020-08-25:
Hmm, maybe try running `clean_rebuid.cmd` from the root directory, that should rebuild everything.
Do commit any changes made in the files present in the Blocks directory like in the AirLib folder before running though

AOS55, 2020-08-25:
Great thanks that seems to have resolved it back to linking problems probably due to decleration again.

rajat2004, 2020-08-25:
Nice! Still have no idea why VS decided to remove the header file
Side note, this does somewhat bring into mind that GitHub issues might not be the best way to communicate, especially on smaller problems. Maybe something for future

AOS55, 2020-08-25:
Yeah not sure its unusual. 
Yes certainly, I'm surprised there isn't a microsoft/AirSim slack setup or equivalent for more direct communication. As you said something for the future.

rajat2004, 2020-08-25:
There is a Facebook group, but I myself haven't used it since a long time.
AP uses Discord (earlier Gitter) and has a forum as well, similar with PX4. Maybe after there is more activity and members. Having something like this would definitely reduce the number of issues also, since more confirmed issues will be created rather than duplicate and support ones

AOS55, 2020-08-25:
Yes I agree thats a good idea, the documentation for AirSim would need to reflect this too as there is not an obvious point of contact when onboarding new users/developers with the application. 

Also after commenting out the linking errors in the application I have managed to compile through to the blocks environment 😀.

rajat2004, 2020-08-25:
Awesome, now comes the probably more difficult part of modelling, physics and making sure everything works!
Having a more direct method of communication will also increase community interaction a lot. Which platform to select, etc will have to be decided by the maintainers though, might require collecting some opinions from the users

AOS55, 2020-08-25:
Yes I agree, would you like to open up a communication issue, I'm not sure if you can do a poll on github?

rajat2004, 2020-08-25:
Don't think I'm the best person to open such a issue, a maintainer should do this probably. Github doesn't have polls AFAIK

AOS55, 2020-08-25:
Yes that makes sense, a moderator or project manager sort of thing. Perhaps if you know anyone you could `@` them into this issue so they can set it up. Ok sure you could always make the proposed solution as a comment and get people to vote 👍  or 👎 on the comment.

rajat2004, 2020-08-25:
@madratman @saihv @ironclownfish @jonyMarino Do you think having another (more chat & voice type) communication platform will be worthwhile? This will definitely increase community interaction, and will probably reduce support issues.
Just wanted to know your opinions, and definitely don't want to increase your (already heavy) workload too much.
Looking forward to hearing from you!

rajat2004, 2020-08-26:
@AOS55 AirLib is actually not compiling for me, getting some undeclared identifier errors for the function declarations which have been commented out. Here's a Travis build - https://travis-ci.com/github/rajat2004/AirSim/builds/181419292, even the Windows build fails.
Did some files not get committed when commenting out the API methods?

Would also recommend activating Travis on your Github repositories (atleast Airsim), that has saved me a few times when making some changes which don't compile on Windows, or breaking something else. (Side note, I did break the Windows build in my first contribution to Airsim (and others later on also), that was a big motivation to get Travis setup properly on Airsim)

AOS55, 2020-08-26:
@rajat2004 hmm ok, this is interesting could you please send me the identifier errors you are getting I will cross-check them on my local machine, I just ran update_to_git.bat and had no further changes and I also ran update_from_git.bat in Blocks and it still compiles ok on my end. A couple of thoughts:
- Are you using UE4.25.3 or UE4.24.3? I have been using 4.25.3 I don't know if this would effect the build or travis. 
- Did you try rebuilding from the root directory as that seemed to be what caused mine to compile in the first instance?

Nice Idea, I have setup [Travis with my branch](https://travis-ci.com/github/AOS55/AirSim-1/builds/181431701), and it also appears to fail. Looking at the job log it has the following on line 1457. I wonder if this is unity related or just what the [.travis.yml](https://github.com/AOS55/AirSim-1/blob/FixedWing/.travis.yml#L37) script should print to the command line. 

`The command "if [[ "$TRAVIS_OS_NAME" == "linux" ]]; then ./setup.sh; ./build.sh || travis_terminate 1; echo "Starting Unity Build!"; (cd Unity && ./build.sh || travis_terminate 1); elif [[ "$TRAVIS_OS_NAME" == "windows" ]]; then export PATH=$MSBUILD_PATH:$PATH_FIX:$PATH; cmd.exe //C 'C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Auxiliary\Build\vcvarsall.bat' amd64 '&&' build.cmd '&&' cd Unity '&&' build.cmd; elif  [[ "$TRAVIS_OS_NAME" == "osx" ]]; then ./setup.sh; ./build.sh || travis_terminate 1; echo "Starting Unity Build!"; (cd Unity && ./build.sh || travis_terminate 1); fi" exited with 1.`

rajat2004, 2020-08-26:
Very strange that it works on your machine, these are the errors I'm getting - 
```
/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp:514:9: error: use of undeclared identifier 'safetyCheckVelocity'
    if (safetyCheckVelocity(Vector3r(vx, vy, vz)))
        ^
/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp:526:9: error: use of undeclared identifier 'safetyCheckDestination'
    if (safetyCheckDestination(dest))
        ^
/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp:538:9: error: use of undeclared identifier 'safetyCheckVelocity'
    if (safetyCheckVelocity(getVelocity()))
        ^
/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp:562:9: error: use of undeclared identifier 'safetyCheckVelocity'
    if (safetyCheckVelocity(getVelocity()))
        ^
[ 80%] Building CXX object AirLib/CMakeFiles/AirLib.dir/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/multirotor/api/MultirotorApiBase.cpp.o
[ 81%] Building CXX object AirLib/CMakeFiles/AirLib.dir/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp.o
4 errors generated.
AirLib/CMakeFiles/AirLib.dir/build.make:230: recipe for target 'AirLib/CMakeFiles/AirLib.dir/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp.o' failed
make[2]: *** [AirLib/CMakeFiles/AirLib.dir/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 83%] Linking CXX executable ../../output/bin/MavLinkTest
[ 83%] Built target MavLinkTest
/home/rajat/Github/AirSim-FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingRpcLibServer.cpp:116:45: error: no member named 'moveByManual' in 'msr::airlib::FixedWingApiBase'
        return getVehicleApi(vehicle_name)->moveByManual(vx_max, vy_max, z_min, duration);
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^
1 error generated.
```

AirLib compilation doesn't depend on UE, Travis also doesn't have UE, it just tests whether AirLib, ROS wrapper compiles, and there's the recently added Azure Pipelines CI which compiles the entire Blocks project on Ubuntu and Windows and packages it.

Yup, I'm rebuilding from the root directory, `./clean.sh & ./build.sh`.

That's just Travis printing the entire command which failed to run, the errors can be seen above it, starting from [here](https://travis-ci.com/github/rajat2004/AirSim/jobs/377806711#L1420)

AOS55, 2020-08-26:
Ok yeah, I was just looking at the code on my machine and I have those if statements at the lines the compiler called out still there, given I commented them out in the header file I would have expected an undeclared header error. I have commented them out now in the [CPP file](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingApiBase.cpp#L514). Still waiting for Travis.

AOS55, 2020-08-26:
Got some further travis build errors in [FixedWingRpcLibServer.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/src/vehicles/fixedwing/api/FixedWingRpcLibServer.cpp) commented out the moveByManual Function.

rajat2004, 2020-08-26:
@AOS55 this was a problem with the CMake config in ROS wrapper, I've opened #2978 which fixes this. The reason's described there, and as to why it was missed, looks like not many people have named the folder something other than `AirSim` :)

AOS55, 2020-08-26:
Thanks, yep, I should have left it as was, good inadvertent check though. Made equivalent commit to FixedWing branch. 

I met with my PhD supervisor earlier and we had a couple of thoughts on what we need to continue developing Fixed Wing Airsim, let me know what you think should be our priorities over the next week and what we have missed:
- Have a runnable fixedwing model in AirSim, I have added a [blueprint](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Content/Blueprints/BP_FixedWingPawn.uasset) for a fixedWingPawn. I also have a 3D model of one of our FW UAVs I should be able to implement this as a material and mesh, just working out how to convert in maya. 
- I tried to run the FW UAV by writing in `settings.json` -> `"VehicleType": "arduplane"` but received  in UE4 `There were no compatible vehicles created for current SimMode! Check your settings.json.`. I'm not sure if that was an error in the json file or if it is related to the above issue or something deeper. 
- We have a validated F-16 flight physics model from [Heffely & Jewell](https://books.google.co.uk/books?id=lvhcCgAAQBAJ&source=gbs_similarbooks) in Matlab that can be used to test the fixedwing physics hpp file. I think it might be a good idea to setup some test cases so I can validate the flight physics functions. I'm not sure if a similar sort of thing was done to validate for the multirotor model?
- Validate and tune some of the controllers for the Ardupilot Mavlink API I'm not sure how much of this is already done in Arduplane? This will probably also be related to writing a python API but that is probably further into the horizon.

rajat2004, 2020-08-26:
Nice! I'll try out the latest commits and have a look at the asset, but I don't have much experience with building models, etc so probably won't be of much help here.
Will test the settings issue too.

I'm not sure how the multirotor model was tested and validated, the maintainers would have a better idea since they wrote this :)
There are many very experienced people in the ArduPilot community who would know about building and validating the physics, might be useful to have a talk with them. I remember there was recently quite a bit of work done on XPlane and RealFlight which are used extensively for Plane simulations

rajat2004, 2020-08-26:
Ok, so for the settings, I think there should be a new SimMode added for Plane, so the options would become like `Multirotor`, `Car`, `ComputerVision` & the new `FixedWing` (or something). This will require changes in a few places, maybe keep following places where `"Multirotor"` or others are used to compare the SimMode name, this commit can also be helpful - https://github.com/microsoft/AirSim/pull/2861/commits/faee309415fde2d946ad3cefdaafa9e7fb251f60, I've tried to make SimMode names constants like `kVehicleType*`.
For a shortcut right now, you could just add ArduPlane in https://github.com/microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/Vehicles/Multirotor/SimModeWorldMultiRotor.cpp#L95

rajat2004, 2020-08-28:
@AOS55 Just tried out the latest commits, the fixedwing model looks awesome :heart_eyes: 
For someone following, use UE 4.25, that's the version the assets are saved with. Some pics -
[image]
[image]

AOS55, 2020-08-29:
I have aimed to add the fixedwing class to the [AirSimSettings.hpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/common/AirSimSettings.hpp) file. `arduplane` should be set as the default FixedWing variable, referenced to `static constexpr char const * kVehicleTypeArduPlane = "arduplane"`. [This commit](https://github.com/AOS55/AirSim-1/commit/693b25e758c57f31fa5b15e3ef61c772df2fd1c4) shows the changes I made to the repo. In particular I have added variable `simmode_name == "fixedwing"` to the `createSimMode` method in [simHUD.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/SimHUD/SimHUD.cpp#L306). 

When I run airsim in unreal with `"SimMode": "FixedWing"` I get an error as follows:
`Error at startup: invalid map<K, T> key`. 

UE4Editor-Win64-DebugGame.exe then logs exception thrown in VisualStudio and produces a breakpoint @ line 73 in [AirSimGameMode.cpp](https://github.com/AOS55/AirSim-1/blob/FixedWing/Unreal/Plugins/AirSim/Source/AirSimGameMode.cpp#L73). I think this is because it can't StartPlay as it doesn't know where the files are. I'm guessing I have not defined the _keys_ at some point, but not sure where this is? Could it be to do with the `VehicleSettings` method in [AirSimSettings.hpp line 807](https://github.com/AOS55/AirSim-1/blob/FixedWing/AirLib/include/common/AirSimSettings.hpp#L807)?

Not sure if this has come up before?

rajat2004, 2020-08-29:
Opened https://github.com/AOS55/AirSim-1/pull/5 which fixes a compilation error and the pawn path aka Invalid Key error
After this the ArduPilot Plane backend start working!
```
[2020.08.29-18.05.26:678][971]LogTemp: Using UDP port 9003, local IP 127.0.0.1, remote IP 127.0.0.1 for sending sensor data
Using UDP port 9003, local IP 127.0.0.1, remote IP 127.0.0.1 for sending sensor data
[2020.08.29-18.05.26:678][971]LogTemp: Using UDP port 9002 for receiving rotor power
Using UDP port 9002 for receiving rotor power
Error while receiving aircraft control data - ErrorNo: -1
Error while receiving aircraft control data - ErrorNo: -1
...
```

However I'm seeing some problems with the assets, need to look into them-
```
[2020.08.29-18.05.26:125][971]LogActor: Warning: BP_CameraDirector_C /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap.FlyingExampleMap:PersistentLevel.CameraDirector has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
1001[2020.08.29-18.05.26:128][971]LogLinker: Warning: While loading '../../../../../Github/AirSim-FixedWing/Unreal/Environments/Blocks/Plugins/AirSim/Content/Models/FixedWingUAV/BUDDI3D_1.uasset' failed to load '/AirSim/Models/FixedWingUAV/Budee': Can't find file.
1001[2020.08.29-18.05.26:129][971]LogLinker: Warning: While loading '../../../../../Github/AirSim-FixedWing/Unreal/Environments/Blocks/Plugins/AirSim/Content/Models/FixedWingUAV/BUDDI3D_1.uasset' failed to load '/AirSim/Models/FixedWingUAV/phong4': Can't find file.
[2020.08.29-18.05.26:135][971]LoadErrors: Failed to load /AirSim/Models/FixedWingUAV/Budee.Budee Referenced by FbxStaticMeshImportData_0
[2020.08.29-18.05.26:135][971]LoadErrors: Failed to load /AirSim/Models/FixedWingUAV/phong4.phong4 Referenced by BUDDI3D_1
```

AOS55, 2020-08-29:
Thanks a lot, merged the commit back onto fixedwing branch.
I see the asset problems, and it now proceeds directly to the breakpoint, Ill start looking into what is not there.
Did you get the ArduPilot Plane backend output from the Visual Studio command line output?
Yes I agree reducing the PawnPath names to the same structure as teh vehicle types would make life simpler.

rajat2004, 2020-08-29:
I'm on Linux :) So all the output comes on the terminal itself from where UE is launched, makes life simpler. 
But more painful for debugging, VS is better for that. Otuptu should be possible from there, but not sure.

Asset problems a PITA, was poking around with some Weather problems today

AOS55, 2020-08-29:
Uh ok nice will probably switch back to linux at some point. That makes sense Ill take a closer look what was your `settings.json` file set as that might be it? I have this:
{
   "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
   "SettingsVersion": 1.2,
   "SimMode": "FixedWing"
}

Haha, ok sure that sounds like fun.

rajat2004, 2020-08-29:
Yeah sorry, my `settings.json` -
```
{
  "SettingsVersion": 1.2,
  "SimMode": "FixedWing",
  "Vehicles": {
    "Plane": {
        "VehicleType": "ArduPlane",
        "UseSerial": false,
        "LocalHostIp": "127.0.0.1",
        "UdpIp": "127.0.0.1",
        "UdpPort": 9003,
        "ControlPort": 9002
    }
  }
}
```

Wouldn't call fun, ran my system at full throttle though, packaging multiple times does that. Issue #2988 if interested
I progressed (more aptly regressed) from crashing binary to when API called, then when pressing Play and finally starting up itself plus compilation, in an infinite loop

rajat2004, 2020-08-29:
Also, just remembered this, you'll want to use `SteppableClock` for AP, so need to add the simmode here - https://github.com/microsoft/AirSim/blob/master/AirLib/include/common/AirSimSettings.hpp#L1116
Not sure about PX4, but AP definitely would like SteppableClock

AOS55, 2020-08-29:
Great thanks, just ran with that `settings.json` and got the same in the VS output. 

Yes Im sure Ill take a look, and likewise the BLUEPRINTS are also new to me. It looks very frustrating Ill feedback if I can think of anything while debugging the FixedWing blueprints. 

I was thinking about that today actually, yeah its certainly going to be needed at some point. Cool Ill add kVehicleTypePX4Plane into there and an or `"FixedWing"` in the if statement.

AOS55, 2020-08-30:
Ive made some more progress with the errors and `ArduPilot/ArduPlane`. I renamed and updated the aircraft mesh within ardupilot on the fixedwing side this resolved most of the can't find file errors discussed earlier. I still have some of the earlier weather can't find errors described above as a current issue. 

The following is the final outpu with ErrorNo: 10040, an error in red.
```[2020.08.30-09.41.01:179][159]LogTemp: Plane
[2020.08.30-09.41.01:184][159]PIE: Server logged in
[2020.08.30-09.41.01:186][159]LogTemp: Error while receiving aircraft control data - ErrorNo: 10040
[2020.08.30-09.41.01:186][159]PIE: Play in editor total start time 0.539 seconds.

I have updated the Arduplane SITL on the [FixedWing branch](https://github.com/AOS55/ardupilot) with an arduplane implementation. This can be compiled and run with WAF by typing:
`sim_vehicle.py -v ArduPlane -f airsim-plane --console --map` within the ardupilot directory. 

Any thoughts on the error im getting `Error while receiving aircraft control data - ErrorNo: 10040`. Im not sure if the correct ports are exposed on the ArduPilot SITL side as it produces the following on start:

```SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--map" "--console" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501"
RiTW: Starting ArduPlane : /home/quessy/ardupilot/build/sitl/bin/arduplane -S --model airsim-plane --speedup 1 --defaults /home/quessy/ardupilot/Tools/autotest/default_params/plane.parm,/home/quessy/ardupilot/Tools/autotest/default_params/airsim-plane.parm -I0
xterm: cannot load font "10x20"
xterm: cannot load font "-Misc-Fixed-bold-R-*-*-13-120-75-75-C-60-ISO8859-1"
xterm: cannot load font "-Misc-Fixed-medium-R-*-*-13-120-75-75-C-120-ISO10646-1"
xterm: cannot load font "-Misc-Fixed-bold-R-*-*-13-120-75-75-C-120-ISO10646-1"
xterm: cannot load font "-misc-fixed-medium-r-normal--20-200-75-75-c-100-iso10646-1"
Connect tcp:127.0.0.1:5760 source_system=255

Not sure if it is connecting to 14550 rather than 9002, 9003 or is 14550 the Xming graphical port?

rajat2004, 2020-08-30:
Awesome, thanks for the asset fix! Will try out the new commits and the AP branch soon.
From a brief search, the error code 10040 seems to indicate that the packet sizes are not matching. The packet sent by AP is defined here - https://github.com/ArduPilot/ardupilot/compare/master...AOS55:FixedWing#diff-ce0d1595ca9423d29853943664d2a43aR70 and has an extra `tla` field which is not defined on the AirSim side, so the buffer size is too small and hence the error. Fixing that should solve the problem, but need to test

AOS55, 2020-08-30:
Cool that was actually it commented out that line and then the asset loaded into unreal, see image below. I have set the axis slightly wrong on the static mesh so the front is currently the side, think I need to move the camera back a bit as the defaultUAV model is much larger than the default quadcopter.  

Im going to start analyzing the physics model and I will add in some sensible parameters for the aircraft. [This skywalker paper]( http://folk.ntnu.no/richahan/Publications/2018_Gryte%20et%20al_ICUAS.pdf) offers quite a nice american derivative non-linear model for the skywalker UAV that will be fine for now. 

Im also going to have a think about setting up some kind of hpp or python output test file that can work out whether or not the model is correctly implemented, perhaps by running some quasi-sim based flight test. Or there might be something more direct by running the simulation hpp files in isolation and giving them a datum input & output. 

[image]

AOS55, 2020-08-30:
Just realised that the skywalker x8 is a blended wing body design and doesn't have seperate aileron elevator and rudder controls. I might implement the aircraft dynamics as described for now and artificially add in control surfaces just to see how it performs. 
There are other models I might take a look at there is a nice collection in [Heffely and Jewell](https://strives-uploads-prod.s3.us-gov-west-1.amazonaws.com/19730003312/19730003312.pdf?AWSAccessKeyId=AKIASEVSKC45ZTTM42XZ&Expires=1598813124&Signature=bYaWe8Rnx2zjZI6LwQI9UHRfYEE%3D) despite it being a pain to read through. [Aircraft Control and Simulation](https://www.amazon.co.uk/Aircraft-Control-Simulation-Dynamics-Autonomous/dp/1118870980/ref=pd_lpo_14_t_0/262-5225549-2222138?_encoding=UTF8&pd_rd_i=1118870980&pd_rd_r=491f98da-59de-4052-a0fc-f99e171eed28&pd_rd_w=QtwWX&pd_rd_wg=UJ3y8&pf_rd_p=7b8e3b03-1439-4489-abd4-4a138cf4eca6&pf_rd_r=2A2ZBBC7PYY9WM3100S2&psc=1&refRID=2A2ZBBC7PYY9WM3100S2) also has an F16 fixedwing model. It might be a case of scaling back a generic fixedwing dynamic model for a generic-UAV model if setup in a conventional configuration. There is quite a lot out there on similtude for dynamics.

samuelctabor, 2020-08-30:
An option you might want to consider would be using the same coefficients as the ArduPilot built-in fixed-wing simulator coefficients defined here https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_Plane.h

This would have the advantage that the ArduPlane tuning would be correct from the start, and if it flies well using these that would be basic validation of physics implementation.

AOS55, 2020-08-30:
Cool thanks  a lot, didn't realised that existed I will just use that makes life much easier. Ill set that up as the genericAircraft on the airsim side too.

AOS55, 2020-08-31:
I've added the fixedwing aircraft from [SIM_Plane.h](https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_Plane.h) along with the model from the Skywalker X8 paper. Im going to start looking at running some tests today and using ardupilot in some more detail.

AOS55, 2020-09-01:
Does anyone know how to change the initialization altitude in the blocks environment. I have tried initializing the aircraft with the following JSON but it keeps starting the aircraft on the ground:
```
{
  "SettingsVersion": 1.2,
  "SimMode": "FixedWing",
  "Vehicles": {
    "Plane": {
      "VehicleType": "ArduPlane",
      "UseSerial": false,
      "LocalHostIp": "127.0.0.1",
      "UdpIp": "127.0.0.1",
      "UdpPort": 9003,
      "ControlPort": 9002,
      "OriginGeopoint": {
        "Latitude": 47.641468,
        "Longitude": -122.140165,
        "Altitude": -800
      }
    }

  }
}
```

I thought this would start at 800m above the ground but is still starting on the floor.

rajat2004, 2020-09-01:
`OriginGeopoint` goes outside the `Vehicles` section, along with `SimMode`, `SettingsVersion`

AOS55, 2020-09-01:
Great thanks, it now looks sort of like it is dropping in but immediately ends on the floor, do you set the fps from the settings.json as the mavlink console says `FPS avg=333.33` Im wondering if its running so fast that you cant see it descending. Or do you know of a method to get a log from ardupilot or airsim

rajat2004, 2020-09-01:
The 333.33 Hz FPS is the physics rate, not the visual one. FastPhysicsEngine runs at that fixed rate, cna be slowed using `ClockSpeed`. I'm not sure what you mean by dropping immediately, the Alt reported by Mavproxy windows goes to 0? A screen capture would be nice, will try it out soon.
AP stores logs in a folder called `logs`, in the current directory

AOS55, 2020-09-01:
Ok that makes sense, yes sorry should have been clearer as the object renders into unreal it looks like the tail comes down and then it lies flat on the ground but that could just be it initializing. I have a screen grab here, great thanks.  

I can't see the logs folder is it in the ardupilot directory or the airsim directory, I see `mav.tlog` but that seems to be a binary or is unreadable with `cat mav.tlog`.

I also notice that Ardupilot calls `APM: Ground start complete` in the console do you know what that does or is it just a sensor initialization command?

[image]

rajat2004, 2020-09-01:
Okay, I think the logs haven't been created since the vehicle isn't armed, you can set `LOG_DISARMED` to 1, that will generate the logs without arming the vehicle. Also, the logs are binary files, you can explore them using Mision Planner or `Mavexplorer.py`

Probably sensor initialization, I haven't played too much with Plane so not sure

AOS55, 2020-09-01:
Cool, got that working added `LOG_DISARMED = 1` into [airsim_plane.parm](https://github.com/AOS55/ardupilot/commit/afa212a6a87260a925fae848e70cd0963684f354), it seems to be starting on the ground in a vertical attitude from the logs:
[image]. 

Ok sure. 

Im on WSL 1

rajat2004, 2020-09-01:
You could also uncomment the `#if 0` & `#endif` [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_AirSim.cpp#L311-L364), that'll log the raw data sent by AirSim as well in separate messages. It might be better to discuss about the physics and log analysis with the much more knowledgable folks over at AP's Discord server or the forums.

Hmm, haven't had any problems with WSL1 when I tested, maybe something's changed, does it reconnect or stay disconnected?

AOS55, 2020-09-01:
Hmm ok, uncommented them now and removed `'` as this was throwing a warning, thats a good idea Ill send a message in general. I can't seem to see where the ASM1 log message appears

I think it reconnects not sure though this is what typing `messages` into `MAVExplorer.py` gives me:

```
MAV> 2020-09-01 15:16:08 Throttle failsafe off
2020-09-01 15:16:08 ArduPlane V4.1.0dev (1ecc1618)
2020-09-01 15:16:08 5f2e394f13010c736f29b85d5f312ef0
2020-09-01 15:16:08 Param space used: 260/3840
2020-09-01 15:16:08 RC Protocol: SITL
2020-09-01 15:16:08 New mission
2020-09-01 15:16:08 New rally
2020-09-01 15:16:10 Airspeed 1 calibrated
2020-09-01 15:16:14 GPS 1: detected as u-blox at 230400 baud
2020-09-01 15:16:15 EKF2 IMU0 initial yaw alignment complete
2020-09-01 15:16:15 EKF2 IMU1 initial yaw alignment complete
2020-09-01 15:16:20 EKF2 IMU0 tilt alignment complete
2020-09-01 15:16:20 EKF2 IMU1 tilt alignment complete
2020-09-01 15:16:25 EKF2 IMU0 origin set
2020-09-01 15:16:25 EKF2 IMU1 origin set
2020-09-01 15:16:51 EKF2 IMU0 is using GPS
2020-09-01 15:16:51 EKF2 IMU1 is using GPS
```

rajat2004, 2020-09-03:
@AOS55 Just tried out the latest commits, things look okay to me from the logs, noticed that there were some problems with the logs, opened https://github.com/AOS55/ardupilot/pull/1 which enables the logging of Airsim data
Here's a log file with the ASM1, ASM2 messages - https://drive.google.com/file/d/1ZhiCmiXtN6-KAdYzklJlkL6k6HlzI5tN/view?usp=sharing

AOS55, 2020-09-03:
Great thanks a lot Ive commited that back to master, I also added support for tla in the message packet and a very simple linear physics implementation for this. 

Thanks how did you view the log file, was there a specific command in MAVExplorer.py?

[65 more replies]