microsoft/AirSim #2957: OptionsMenu.uasset was saved with a newer custom version than the current

GimpelZhang, 2020-08-21:
Hi all, 

I had an issue on the weather option function during the simulation. 

When I pressed F10, the following errors came up:

```bash
[2020.08.21-01.46.36:854][522]LogLinker: Error: Package ../../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/HUDAssets/OptionsMenu.uasset was saved with a newer custom version than the current. Tag 375EC13C06E448FBB50084F0262A717E Name 'Dev-Core' PackageVersion 4  MaxExpected 3
[2020.08.21-01.46.36:854][522]LogLinker: Error: Package ../../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/HUDAssets/OptionsMenu.uasset was saved with a newer custom version than the current. Tag 375EC13C06E448FBB50084F0262A717E Name 'Dev-Core' PackageVersion 4  MaxExpected 3
[2020.08.21-01.46.36:854][522]LogUObjectGlobals: Warning: Failed to find object 'Class /AirSim/HUDAssets/OptionsMenu.OptionsMenu_C'
[2020.08.21-01.46.36:855][522]LogLinker: Error: Package ../../../../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Content/HUDAssets/OptionsMenu.uasset was saved with a newer custom version than the current. Tag 375EC13C06E448FBB50084F0262A717E Name 'Dev-Core' PackageVersion 4  MaxExpected 3
[2020.08.21-01.46.36:855][522]LogTemp: Warning: Warning, WeatherAPI could not load weather widget!

```

* my settings:

```
ubuntu 16.04
UE 4.24.3
AirSim: the latest Master branch until the commit on Aug 05
```

I suppose that this PR #2877 has something to do with this issue, since after I changed back the files mentioned in this PR, the errors didn't occur anymore and the weather options worked. 

**Please don't bother if this is about new releases aiming on unreal 4.25.**

xxEoD2242, 2020-10-31:
@GimpelZhang

I had the same issues. What fixed this for me (and it's really more of a work around) was to remove the `.uasset` files from the HUDAssets folder in the plugin folder for AirSim. I'm pretty sure this will break the UI but it will allow you to cook a game.

I can give step by step directions if you need it. FYI, I am doing this on Ubuntu 20.3.4, Unreal 4.24 and using the latest pull of AirSim. I also did this on AWS so my compile times were slower.

GimpelZhang, 2020-11-01:
> @GimpelZhang
> 
> I had the same issues. What fixed this for me (and it's really more of a work around) was to remove the `.uasset` files from the HUDAssets folder in the plugin folder for AirSim that are causing the errors. Missing files only generate a warning. I'm pretty sure this will break the UI but it will allow you to cook a game.
> 
> I can give step by step directions if you need it. FYI, I am doing this on Ubuntu 20.3.4, Unreal 4.24 and using the latest pull of AirSim. I also did this on AWS so my compile times were slower.

Thanks a lot! 

I will give it a try when I have chance.

jonyMarino, 2021-08-19:
Now we are using v4.25
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!

closed 2021-08-19 by jonyMarino