microsoft/AirSim #2964: Unable to accept TCP connection from PX4 running on docker

Cypre55, 2020-08-24:
Hi,
I'm new to aerial robotics, so please correct me if I use the wrong terms anywhere.

I am trying to get PX4 to connect to airsim. So, I first changed the settings accordingly.
```json
 {
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "UseSerial": false,
            "UseTcp": true,
            "TcpPort": 4560,
            "ControlPort": 14580,
            "params": {
                "NAV_RCL_ACT": 0,
                "NAV_DLL_ACT": 0,
                "LPE_LAT": 47.641468,
                "LPE_LON": -122.140165,
                "COM_OBL_ACT": 1
            }
        }
    }
}
```
To run the docker container I used the following command:
```bash 
docker run -it --privileged \
    -v ~/src/Firmware:/src/firmware:rw \
    -v /tmp/.X11-unix:/tmp/.X11-unix:ro \
    -e DISPLAY=:0 \
    -p 127.0.0.1:4560:4560/tcp \
--name=px px4io/px4-dev-ros-melodic
```

Once in the container, I did
```bash 
cd /src/firmware
make px4_sitl none_iris
```

Everything worked smoothly to the point where PX4 is waiting for airsim to accept the connection on TCP port 4560.

Upon running airsim, encountered the following error:
```
Waitin for TCP connection on port 4560, local IP 127.0.0.1
Accepting TCP socket failed, is another instance running?
TcpClientPort socket bind failed with error:10048
``` 

How can I fix this?

* Airsim version: 1.3.1
* Docker version: 19.03.12
* Docker image: px4io/px4-dev-ros-melodic
* Windows version: 2004 (OS Build 19041.450)
* Linux distribution on WSL 2: Ubuntu 20.04 LTS

wuhanstudio, 2021-03-24:
Hi, 

Is the docker running under WSL1? Usually it's better to use WSL2, which has a full Linux kernel allowing Linux containers to run natively without emulation.

k31415, 2023-02-14:
Hi,

did you solve this issue? I have the same problem...

mehmetbozhaydar, 2025-01-28:
@Cypre55 Hi,

Did you solve this issue? I have the same problem... Please help!