microsoft/AirSim #2965: msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'drone0' is not available. This could either because this is simulation-only API or this vehicle does not exist.

XueZhihan, 2020-08-24:
I'm getting this error when trying airsim1.2.4 and 1.2.8.

```
Traceback (most recent call last):
  File "C:/SC/AirSim SC/PEDRA/main.py", line 99, in <module>
    eval(name)
  File "<string>", line 1, in <module>
  File "C:\SC\AirSim SC\PEDRA\algorithms\DeepQLearning.py", line 21, in DeepQLearning
    client, old_posit, initZ = connect_drone(ip_address=cfg.ip_address, phase=cfg.mode, num_agents=cfg.num_agents)
  File "C:\SC\AirSim SC\PEDRA\aux_functions.py", line 300, in connect_drone
    client.enableApiControl(True, name_agent)
  File "C:\tools\IDEs\Anaconda3\lib\site-packages\airsim\client.py", line 37, in enableApiControl
    return self.client.call('enableApiControl', is_enabled, vehicle_name)
  File "C:\tools\IDEs\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\tools\IDEs\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'drone0' is not available. This could either because this is simulation-only API or this vehicle does not exist.
```
And my setting.json in ~\Documents\Airsim is looked like this:
```
{
    "SettingsVersion": 1.2,
    "LocalHostIp": "127.0.0.5",
    "SimMode": "Multirotor",
    "ClockSpeed": 20,
    "ViewMode": "NoDisplay",
    "PawnPaths": {
        "DefaultQuadrotor": {
            "PawnBP": " Class'/AirSim/Blueprints/BP_DJIMavic.BP_DJIMavic_C' "
        }
    },
    "Vehicles": {
        "drone0": {
            "VehicleType": "SimpleFlight",
            "X": 0.0,
            "Y": 0.0,
            "Z": 0,
            "Yaw": 180.0
        }
    },
    "CameraDefaults": {
        "CaptureSettings": [
            {
                "ImageType": 0,
                "Width": 320,
                "Height": 180,
                "FOV_Degrees": 80
            },
            {
                "ImageType": 3,
                "Width": 320,
                "Height": 180,
                "FOV_Degrees": 80
            }
        ]
    }
}
```
Hoping someone can help me out with this. Thank you so much.

OrangeCH3, 2020-09-22:
msgpackrpc.error.RPCError: rpclib: client error C0002: Function 'getHomeGeoPoint' was called with an invalid number of arguments. Expected: 0, got: 1
Hoping someone can help me out with this. Thank you so much.

levicorpus0407, 2021-02-21:
> I'm getting this error when trying airsim1.2.4 and 1.2.8.
>
> ```
> Traceback (most recent call last):
>   File "C:/SC/AirSim SC/PEDRA/main.py", line 99, in <module>
>     eval(name)
>   File "<string>", line [...]

hi, i encountered the same problem. Have you solved this in the end?

BrenoPimentel, 2021-08-28:
Hello everyone
Did you guys solve it?

levicorpus0407, 2021-08-28:
> Hello everyone
> Did you guys solve it?
It’s because the path of my config file contains non-English symbols… I created a new account of my pc and solved the problem. But I’m not sure if you have the same problem.