microsoft/AirSim #2969: How to create video stream like in airsim demo

p3jawors, 2020-08-25:
Hello,

I am looking at getting a video stream picture-in-picture of the rgb camera feed from the multirotor like was done in this airsim demo.
https://www.youtube.com/watch?v=-WfTr1-OBGQ

So far I have found the `opencv_show.py` example, but this creates a separate window with an extremely low FPS count. How would I go about recreating what was in the video above with a camera feed in the unreal editor with the higher fps video?

I have read through this post https://github.com/microsoft/AirSim/issues/892 with a similar issue, but I would like to keep the video stream within unreal, as in the video above.

Cheers

rajat2004, 2020-08-26:
You're looking for Subwindows, they can be activated by pressing 1,2,3
For changing the camera views, see https://microsoft.github.io/AirSim/settings/#subwindows

p3jawors, 2020-08-26:
Wow was I ever overcomplicating it haha. Thanks for the quick response.

closed 2020-08-26 by p3jawors