microsoft/AirSim #2970: Sending messages from /cmd_vel to an AirSim topic

nmuchaxo, 2020-08-25:
Hello, I've been trying to subscribe to /cmd_vel (type: geometry_msgs/Twist) to publish info to the topic /airsim_node/drone_1/vel_cmd_body_frame to move the drone in the sim and I thought both had the same msg type, but apparently not. The type of the last topic is airsim_ros_pkgs/VelCmd and it just has Twist inside it.

Inititally my code was like this and when I run it I get "WARN Could not process inbound connection: topic types do not match" which makes sense, however after discovering the types are different I don't know what imports I have to do in order to pass from Twist to VelCmd. Can someone help me?

`
#!/usr/bin/env python                                                                             

import rospy
import roslib
import tf
from std_msgs.msg import String
from geometry_msgs.msg import Twist

def callback(msg):
    pub = rospy.Publisher('/airsim_node/drone_1/vel_cmd_body_frame', Twist, queue_size=10)
    pub.publish(msg)
    rospy.loginfo("Publishing: [%f, %f, %f]" %(msg.linear.x, msg.linear.y, msg.linear.z))

def listener():
    global msg
    rospy.init_node('cmdvel_listener', anonymous=True)
    msg = Twist
    rospy.Subscriber("/cmd_vel", Twist, callback)
    rospy.spin()  

if __name__ == '__main__':
    listener()  `

GimpelZhang, 2020-08-26:
Obviously you need to import the VelCmd type into your python code. 

I guess what you are trying to do is to translate the subscribed /cmd_vel command to VelCmd, and then to publish it. Then you need to publish a message in VelCmd type. 

Here is a script I wrote aiming to do something else, but I hope it can help a little to your problem: [send VelCmd command from keyboard](https://github.com/GimpelZhang/airsim_ros/blob/master/airsim_car_teleop/nodes/airsim_car_teleop_key)

nmuchaxo, 2020-08-26:
Thanks a lot for replying and giving me an example.

I gave it an initial look and saw the import "from airsim_ros_pkgs.msg import VelCmd" which I had already tried yesterday but I got this message when I try to run the script "ImportError: No module named airsim_ros_pkgs.msg".
Am I missing something else?

GimpelZhang, 2020-08-27:
> Thanks a lot for replying and giving me an example.
> 
> I gave it an initial look and saw the import "from airsim_ros_pkgs.msg import VelCmd" which I had already tried yesterday but I got this error on the console when I try to run the script "ImportError: No module named airsim_ros_pkgs.msg".
> Am I missing something else?

That's probably because you also need to do some work on the package.xml and CMakeLists.txt files in your ros packages, to make the customer defined message type VelCmd work in your project. 

Maybe you can take a look at my package and CMakeList files in both airsim_ros_pkgs and airsim_car_teleop folders.

nmuchaxo, 2020-08-27:
I did look at your CMake, xml and setup files and they seem to be equal to mine. However I still get the "ImportError: No module named msg" error...

GimpelZhang, 2020-08-28:
> I did look at your CMake, xml and setup files and they seem to be equal. However I still get the "ImportError: No module named msg" error...
> Does the folder where the script, the Cmake, setup are... need to be inside the ROS Wrapper folder or the arisim_ros_pkgs?

As you probably have already noticed, that the following blocks of code are mainly about this issue:

from /airsim_ros_pkgs/CMakeLists.txt:

```
add_message_files(
  FILES
  GimbalAngleEulerCmd.msg
  GimbalAngleQuatCmd.msg
  GPSYaw.msg
  VelCmd.msg
  VelCmdGroup.msg
  CarControls.msg
  CarState.msg
  Altimeter.msg
  Environment.msg
)
```
and
```
generate_messages(
  DEPENDENCIES
  std_msgs
  geometry_msgs
  geographic_msgs
)
```
These make sure that the customer msg works in the project. More details: [Creating a ROS msg and srv](http://wiki.ros.org/ROS/Tutorials/CreatingMsgAndSrv#Creating_a_msg)

from /airsim_car_teleop/package.xml:
```
<build_depend>airsim_ros_pkgs</build_depend>
  <exec_depend>airsim_ros_pkgs</exec_depend>
```
This part ensures that your py script project knows where to find the customer msg. 

And, of course, don't forget to run catkin_make.

nmuchaxo, 2020-08-28:
Ok, now I got it to work, kinda! Ended up cleaning the build and devel. The communication is correct and when I do "rostopic echo /airsim_node/drone_1/vel_cmd_world_frame" its getting the data correctly, however when I click "play" the drone falls to the ground, it gives a warning that the collision happened and it won't move from there. 
Do you have any idea why that happens? or why doens't the drone respond to my commands? 
TIA

GimpelZhang, 2020-08-29:
> Ok, now I got it to work, kinda! Ended up cleaning the build and devel. The communication is correct and when I do "rostopic echo /airsim_node/drone_1/vel_cmd_world_frame" its getting the data [...]

The collision warning occurred because the drone hit the ground. 

I'm not quite familiar about the Multirotor SimMode since I have been working on the Car SimMode mainly. 

But for my knowledge it seems that after you click "play", perhaps you should use the "take off" rosservice first to somehow "activate" the drone.

nmuchaxo, 2020-08-31:
I didn't connect the drone to the API and I think I hadn't some things installed (related to AirSim) that were necessary in order for this to work.
Thanks a lot for your help!

closed 2020-09-04 by jonyMarino