microsoft/AirSim #2971: Multi drone API call order in Python yields surprising behavior unless called in a specific way

amaalo2, 2020-08-25:
Using the multi_agent_drone.py as a template to run multiple drone, the suggested code structure is shown below. 

` airsim.wait_key('Press any key to move vehicles')`
`f1 = client.moveToPositionAsync(-5, 5, -10, 5, vehicle_name="Drone1")`
`f2 = client.moveToPositionAsync(5, -5, -10, 5, vehicle_name="Drone2")`
`f1.join()`
`f2.join() ` 

Using this structure, both drones move to their target positions at the same time. 

However, when calling directly the API command without the intermediate variables f1, f2 then the drones would go to their final position one after the other in a sequential manner. This means that the second drone will not move until the first drone has reached its final position.

` airsim.wait_key('Press any key to move vehicles')`
`client.moveToPositionAsync(-5, 5, -10, 5, vehicle_name="Drone1").join()`
`client.moveToPositionAsync(5, -5, -10, 5, vehicle_name="Drone2").join()`

Is this an intended behavior?

xxEoD2242, 2020-08-25:
The method `.join()` is an asynchronous call. That means that when the method is called, the program is suspended until the callback is completed. It's the same as calling `.wait()` on a subprocess in Python, where it waits for the literal `return 0` type callback before exiting the program.

TL:DR Yes, it's functioning as async methods should.

xxEoD2242, 2020-08-25:
How I usually do this is to simply call `.join()` on the last called drone, to ensure that my drones have flown the proper position before executing the next step.

amaalo2, 2020-08-25:
Thank you for confirming this.

closed 2020-08-25 by amaalo2

amaalo2, 2020-08-25:
@xxEoD2242  can you also take a look at https://github.com/microsoft/AirSim/issues/2974 ?