microsoft/AirSim #2976: computer vision mode can not work in custom ureal engine environment

lucafei, 2020-08-26:
hello, guys, i have recently created a custom unreal environment according to the airsim tutorial, after successful creation i tried to use computer vision mode to capture image, but it turned out computer vision mode not supported when i runed the own created unreal environment. Do you know how to fix this or have some advices? Thank you in advance.

clintlombard, 2020-08-26:
I can confirm that Computer Vision mode does work in custom unreal engine environments. I've tested this with UE 4.24.3 and 4.25.3. It might be something in your setup/build of AirSim, please elaborate on those?

jonyMarino, 2022-02-28:
Hi @lucafei! Can you use the `ExternalPhysicsEngine` instead of CV for your use case?